prusa3d/PrusaSlicer #3592: Master branch does not compile

smurfix, 2020-02-01:
```
/src/PrusaSlicer/src/libslic3r/libslic3r.h:239:27: error: format not a string literal and no format arguments [-Werror=format-security]
     int bufflen = snprintf(buffer.data(), INITIAL_LEN - 1, fmt, std::forward<Args>(args)...);
                   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/src/PrusaSlicer/src/libslic3r/libslic3r.h:243:17: error: format not a string literal and no format arguments [-Werror=format-security]
         snprintf(buffer.data(), buffer.size(), fmt, std::forward<Args>(args)...);
         ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

vintagepc, 2020-02-01:
make sure you're using a recent compiler, e.g. gcc-7. There is syntax in the code that isn't supported in gcc-4

smurfix, 2020-02-01:
That's not the problem, I'm using gcc 8.3. The problem is that (a) this function is duplicate anyway, (b) there's no way to tell gcc's C++ that the third argument is the variadic part of snprintf.

I'm preparing a patch.

smurfix, 2020-02-01:
Patch: https://github.com/prusa3d/PrusaSlicer/pull/3594

closed 2020-02-03 by tamasmeszaros