prusa3d/PrusaSlicer #3595: Cisar's vase doesn't slice correctly, creates artefacts

smurfix, 2020-02-01:
### Version
current master, be92e10

### Behavior

Slicing the model of Dominik Cisar's vase with "spiral" option causes artefacts on the letters you embossed on that vase.

See the screenshot, and the 3mf (renamed to .zip for uploading) to reproduce.

[image]
[Vase_Dominik_Cisar.zip](https://github.com/prusa3d/PrusaSlicer/files/4142972/Vase_Dominik_Cisar.zip)

NB: Please consider removing your letters. They are unlikely to look good when I print that vase as a gift …

bubnikv, 2020-02-07:
The mesh slices to multiple contours for some Z heights around the letters.

[image]

[image]

[image]

[image]

I suppose we can extend slicer to do some filtering of the slices, but such operation is always risky as it may change the intent of the designer. Other option is to report an error and give up.

bubnikv, 2020-02-08:
Spiral vase improvements and bugfixes.
Fixes Connecting / expanding Bottom Layers to Vase Perimeter #253
Fixes Slicing error in vase mode #452
Fixes Slicing Issue (Vase Mode, 0.6mm dmr nozzle) #1887
Fixes Top fill pattern isn't used in spiral vase mode #2533
Fixes Cisar's vase doesn't slice correctly, creates artefacts #3595

When the model is sliced, all the contours are newly oriented counter-clockwise (even holes), merged and then only the largest area contour is retained. In perimeter generator, if the largest contour splits into multiple perimeters, newly only the largest area perimeter is retained in spiral vase mode. These two changes solve #3595 and similar.

The infill is newly calculated only for the bottom solid layers if the spiral vase mode is active (removes various unwanted infill along the vase walls), and the last bottom solid layer is switched to a top solid pattern (solves #2533).

The thin walls are newly enforced to be disabled in spiral vase mode, and the "ensure vertical shell wall" is enforced in spiral vase mode to extend the bottom of the vase to the vase hull (fixes #253).

The changes will be released with PrusaSlicer 2.2.0-alpha4. Closing.

bubnikv, 2020-02-08:
Fixed with 4e11552da980f07216f1720f67f539703ed90ed1

closed 2020-02-08 by bubnikv

smurfix, 2020-02-09:
You seem to have forgotten to check in the tests?

bubnikv, 2020-02-09:
Fixed with e7d2fcf0be7a524eb8bf40dd2964539534934e96