prusa3d/PrusaSlicer #3598: Opening STL issues and Wave designs errors, prusa slicer

DavKoun, 2020-02-02:
I have latest Slicer. 
I do printing nonstop large objects on MK3s, many parts glue in to one. 

When i cut in to  21x21x25cm parts and  export stl. files from any program. Very very often i have NO WAY to open large files in Prusa Slicer on my older Macbook pro. (2.4 GHz Quad-Core Intel Core i7,8 GB 1600 MHz DDR3)   Problem is, that exported files are sometimes 5mb stl. and sometimes 300mb....i just need to open in slicer and slice, size already prepared. But prusa slicer doesnt respond and in few seconds......off...    So large files i have to open in Cura. It takes time, but cura makes it in half a minute or so.  You should make option for large files to open and slice.....maybe button with slowing down processor usage for large files, i have no problem to wait minute, or even few when file is large. Almost every large object has at least one of two parts after cut that is large and doesnt work in Prusaslicer. Also would be great to see if prusa slicer processing or is already PROGRAM DOESNT RESPOND.  I prefer prints from Prusa slicer, but this always off me to Cura...

Another big issue in large objects cuted in many pieces, some with WAVE designs. When top layers are slighly shaped.....Try to make object 21x21x25cm like a sea....waves with couple of edges.....YOU WONT EVER PRINT IT WITH PRUSA MK3s  PETG from PRusa slicer....  Your high top wave design will always ruin print..maybe after 1,5days of printing...500g filament.    You need to make improvement of this. Also because when you print nonstop full size objects, you dont wanna make infill more than 5%....and with 3%or 5% you never make those designs printed on MK3s.    (my printer is fine, small objects prints always perfect, i print petg...just that high level of top wave shapes is lottery to make it..)   Maybe solution is to print one piece with 30%-50% infill....but then i will print object 1m large many months with thousands of dollars for filaments, economical non sense even for rich companies....so will be cheaper to make everything handmade, molds and cast......   Solution might be  tests of simple wave designs with edges, around 15-21cm above bed level. And then just add in to gcode some simple supports...simple  Special option for this FULL bed pieces in economy way.....so infill 3% or up to 5%,but some extra automated infill on the top to make print smooth and not a  high risk lottery.....maybe infill will double or triple or 5times more 1cm or 0,5cm under top layers....its easy to make it, but must be tested by PRUSA company...

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-03-15:
please provide models / 3MFs.

bubnikv, 2021-09-22:
no data provided, closing

closed 2021-09-22 by bubnikv