prusa3d/PrusaSlicer #3600: Sequential Printing by height Feature Request

aMoDeRnMoNsTeR, 2020-02-03:
Version
PrusaSlicer 2.1.1

Operating System
Windows 10

3D printer brand
Ender 3

Behavior 
This is a new feature request.

The sequential printing functionality is a bit limited. The type of behavior it has currently is to print by complete object by complete object. The function that I'm looking for is one that is found in Simplify3d, one that I use quite often. Instead of printing an entire object at a time, I'm looking for printing each object a certain amount of height at a time. This helps both with less retraction over open spaces, as well as ensuring less failure possibility. It works by printing all objects up to a specified amount of height (usually the nozzle clearance), then returns to the first object printed and begins printing the same amount of height again on all the objects. If the objects are different heights it continues this process until the first object is finished, then it continues printing the rest of the objects until all objects are finished. Added some pictures to clarify. This allows for printing anything at all at any height without worrying about the nozzle running into anything. 

[image]
[image]
[image]
[image]

worldspawn, 2020-10-12:
Not sure its specifically called out by @aMoDeRnMoNsTeR but prusa slicer won't let you sequentially print if there is any model that is tall enough to collide with the x-axis rail (or it might allow just one, cant remember). This feature would solve that in the most optimal way possible. Sequential printing in it's current form is great when I want to print lots of floor tiles, but when its comes time to print walls progress slows to a crawl.

Tupson444, 2026-09-13:
still wanted