prusa3d/PrusaSlicer #3601: Prusa Is crashing when i am running on another machine.

jayraaj, 2020-02-03:
### Version
PrusaSlicer 2.1.1

### Operating system type + version
Windows 10 64 bit

### Behavior
Actually I wanted the exact same gcode from Prusa command line which we are getting through the GUI so I have modified the PrusaSlicer 2.1.1 and rebuilt on my machine along with it's dependencies using VS 2017 in that i have just minimized UI and made it work as command line now problem is that this is working on mt machine however whenever I am running on another machine it is crashing but not giving any particular error except PrusaSlicer.dll is crashed.

Can any buddy help to solve thiz?

tamasmeszaros, 2020-02-03:
Hi,
if you've built the latest master, I would bet that _libgmp_ and _libmpfr_ is missing which are added as dll-s. I assume you have generated the VS project with cmake. In that case, try to configure the cmake project with `CMAKE_INSTALL_PREFIX` set to your liking and build the `INSTALL` target on VS. It should copy everything that is needed into the install dir you specified and that dir should be almost equivalent to the content of the downloadable zip that we provide -- which is relocatable.

jayraaj, 2020-02-06:
Actually my problem is resolved when I have deleted the folder C:\Users\'User name'\AppData\Roaming\PrusaSlicer and then it is working now.

Thanks.

closed 2020-02-06 by jayraaj