prusa3d/PrusaSlicer #3602: dmg-File corrupted 20200203120000

UltiCa, 2020-02-03:
### Version
Prusa Slicer 2x2x0Beta2

### Operating system type + version
MacOS 10.9.5

### Behavior
when I try to start the PrusaSlicer.app the following message occurs:
"PrusaSlicer.app" ist beschädigt und kann nicht geöffnet werden. Es empfiehlt sich das Objekt in den Papierkorb zu bewegen.

translated:
PrusaSlicer.app is corrupted and cannot be opened. It is recommended to move the object to the trash.

the size of the app-file is:
51.663.655 Byte

bubnikv, 2020-02-03:
Where did you download it from?

UltiCa, 2020-02-03:
https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.2.0-alpha2/PrusaSlicer-2.2.0-alpha2+-201912221330.dmg

UltiCa, 2020-02-03:
Is this a problem with MacOS 10.9.5? I tried Prusa Slicer Version 2.2.0-alpha2 on a Mac of a friend. There it is starting. I hope, Prusa Slicer will be compartible with 10.9.5 :o)

UltiCa, 2020-02-07:
This is still a BUG.

bubnikv, 2020-02-08:
I tried our latest build on 10.9.5, I got the following error:

[image]

Got knows what is going on again. Thanks Apple. Maybe it has something to do with entitlements.

https://github.com/electron/electron-osx-sign/issues/200

steakknife, 2020-02-08:
1. First, try to remove the "downloaded from the scary internets" quarantine attributes from the app entirely: `xattr -dr com.apple.quarantine /path/to/AppName.app`
2. If that doesn't work, <=10.14.x: unsign the app [using a tool like the script here](https://stackoverflow.com/a/51038431) because `codesign` alone is broken. For >=10.15, re-signing the app is required.

bubnikv, 2020-02-20:
Starting with 2.2.0-alpha1 we are using a new build server and new certificates. For some reason signing with these certificates seemed to work, but the signature was rejected by OSX 10.9.5, 10.10 and possibly by some other old OSX versions. We don't know what went wrong.

Anyway, we have a yet new build server freshly set up and quite powerful, and this time the certificates seem to be applied correctly, so they are accepted even by OSX 10.9.5.  I have updated the DMG archive at https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha4, please download https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.2.0-alpha4/PrusaSlicer-2.2.0-alpha4+-202002201411.dmg and test it.

Closing.

closed 2020-02-20 by bubnikv