prusa3d/PrusaSlicer #3607: [Feature Request] Nozzle temperature by height insertion (like how can insert color change or pause print in sliced view)

Photogad, 2020-02-04:
_Is this a new feature request?_
Yes

I would like see ability to insert different nozzle temperatures at different layers, at the same area that you can insert color changes or pause print gcode commands.

Something like this could be used for natively making temperature towers, but more in particular I want to use it to make objects that are matte on the bottom and shiny near the top

bubnikv, 2020-02-04:
You can do that already in any slicer version using the custom layer G-code.
You can do that also in the current PrusaSlicer 2.2.0-alpha2 using the
custom G-code at a given layer set at the G-code preview layer slider
similar to the color change.

rtyr, 2021-02-05:
Answered. It is already possible to do that by adding custom layer g-code (right click on the "+" icon in the preview). There are also some feature requests about adding this option to height range modifiers (for example https://github.com/prusa3d/PrusaSlicer/issues/3635).

Closing.

closed 2021-02-05 by rtyr