prusa3d/PrusaSlicer #3608: Bug: Insert layer pause with Octoprint does not turn nozzle back on

Pun-e, 2020-02-04:
### Version
2.2.0-Alpha2

### Operating system type + version
Windows 10 64

### 3D printer brand / version + firmware version (if known)
Mk2.5S

### Behavior
Inserted pauses into a random cube:
[image]

Print works, everything's fine. Printer pauses, I do my thing. I click resume **but the nozzle does not turn back on** and if not manually corrected, will eventually cause the print to fail

According to[ this forum](https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-others-archive/anyone-tried-the-new-pause-print-gcode-m601-m602/) this should not be an issue caused by octoprint, but I have no way to confirm that for myself.

BeldrothTheGold, 2020-02-07:
Hello! I believe this is a bug with the printer firmware. Changes are being made here: https://github.com/prusa3d/Prusa-Firmware to resolve serial/octoprint related pause bugs.

rtyr, 2020-10-20:
I believe this was already fixed in the firmware. It is not slicer issue. Closing.

closed 2020-10-20 by rtyr