prusa3d/PrusaSlicer #3610: [Feature Request] Analysis of time spent per layer/line type/object

stewcrew14, 2020-02-05:
hello,

Many times when I’m printing I struggle with trying to print within a specific time frame (8 hour shift at work, 4 hours home at night, 12 hours on the weekend), and so I compromise printing speed and printing strong and minimize chance and failure. I know many slicers offer layer views and a color coded images of each line type, but I would like a way to see what % of the total print time is spent on perimeters/infill/top and bottom layers/support/etc. With this information I could modify printing speeds, number of perimeters, infill patterns or density, or anything else to see where I can get the most “bang for my buck” kinda deal. While typing this I realized it can go further then time by line type, it can extend to objects if printing multiple objects or even by layer like octoprint can do. This information should be relatively easy to get since the gcode has the speed and can get the linear length and therefore total time. thanks for reading!

JohnnyDeer, 2020-02-06:
Hi, this issue is duplicate of #2912  , do you agree @stewcrew14 ?

rongith, 2020-03-18:
I'm working on simple layer time preview like that:
[image]

For the moment I think the code need a major rewrite to do what you want. Preview and Time calculation are in two separate classes at the moment and all is computed from g-code twice (Time calculation and Preview).
I think that it will be rewrite but not in a near future. Ideally many statistics per path and layer should be stored for the Preview for easy tweaks but that in not the case.
I decided to work on layer time because the fan speed preview seemed "easy" to implement. And it is as there are only M106 and M107 to check in the Gcode. Layer time preview doesn't really fit in the current code.

I will do a pull request but not expecting Prusa to take it. My code would be scrapped for a major rewrite of GCodePreviewData, GCodeAnalyzer and GCodeTimeEstimator.

bubnikv, 2020-03-18:
>  I will do a pull request but not expecting Prusa to take it. My code
would be scrapped for a major rewrite of GCodePreviewData, GCodeAnalyzer
and GCodeTimeEstimator.

Enrico from our team already started reworking the G-code preview also to
support loading a G-code from PrusaSlicer or 3rd party slicers, so your
conclusion is likely correct.

rtyr, 2021-02-05:
Time per feature was implemented in 2.3.0. 

Time per layer is requested in #2912 #2754.

Closing.

closed 2021-02-05 by rtyr