prusa3d/PrusaSlicer #3611: Slicer produces gaps while using "Detect thin walls"

UltiCa, 2020-02-05:
### Version
2.1.0+ to 2.2.0 alpha2+

### Operating system type + version
MacOS 10.9.5

### 3D printer brand / version + firmware version (if known)
Prusa MK3s with MMU2

### Behavior
The slicer produces gaps which are also printed.
[image]
[image]
[image]

[In the prusa forum there is a more detailed description:](https://forum.prusaprinters.org/forum/prusaslicer/after-slicing-there-is-a-gap-after-printing-too/#post-189756)

The gaps are a result of turning on the "Detect thin walls" option!

#### Project File (.3MF) where problem occurs
[GapProblem.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4158772/GapProblem.3mf.zip)

UltiCa, 2020-02-28:
I have tested 2.2.0 beta
same problems with this issue
[image]

UltiCa, 2020-03-24:
Why is the bug label removed?

The problem still exists. I have tested with the latest versions yesterday.

Version: 2.2.0+ and 2.2.0 rc5
Build: PrusaSlicer-2.2.0+-202003211132
Mac OS Catalina 10.15.3

ProjectCleverWeb, 2022-01-08:
Can confirm this bug still exists

Version: 2.4.0+x64
Build: PrusaSlicer-2.4.0+x64-202112211607
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Big Sur Version 11.6.1 (Build 20G224)
Total RAM size [MB]: 17,180MB

closed 2026-09-29 by prusaslicer-bot[bot]