prusa3d/PrusaSlicer #3614: embedded Items

RusNorthwood, 2020-02-06:
### Version
PrusaSlicer-2.2.0-alpha2+win64-201912221340

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Windows 10 Professional 64-bit
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### MK3S/MMU2S
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* When embedding items in to a multi colour print, you need to press resume print 13 times to get the printer to continue
* select pause to embed at set height
  * _If this is a command-line slicing issue, include the options used_
* click resume print once
* click resume 13 times
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-02-06:
isn't it a printer firmware issue?

BeldrothTheGold, 2020-02-07:
Hello, I agree this is very likely a firmware issue. 

There are known issues with the pause function in the current release of Prusa-Firmware. Especially when printing with Octoprint. 

Please take a look at https://github.com/prusa3d/Prusa-Firmware/issues/1999

bubnikv, 2020-02-08:
> Please take a look at prusa3d/Prusa-Firmware#1999

Would you please close this PrusaSlicer's issue then?

closed 2020-03-15 by bubnikv