prusa3d/PrusaSlicer #3615: slow printing perimeters

3Dietrich, 2020-02-06:
### Version
PrusaSlicer-2.1.1+-201912101508

### Operating system type + version
MacOS Sierra, 10.12.6 (16G2136)

### 3D printer brand / version + firmware version (if known)
Prusa i3Mk3S, Firmware 3.8.1
regarding this Bug: unmodified

### Behavior
_Describe the problem_
I had a 'block' and printed it with 0.6 nozzle, 2 oerimeters, 0% infill, 0 Layer top solid layers so that it becomes a Box (more details in attached zip "wall.." .3mf/gcode)
when printing I found it very slow while printing the walls, running with 150% was not too much. I searched for the speed of this perimeters but nothing did the speed increase.. 
But then I doubled the print object and ..the speed is fine! I dactivated the conditional gcode addition (attached zip "2xwall.." .3mf/gcode)

_Steps needed to reproduce the problem_
compare the estimated printing times (or run it)
* in slicer open the file "wall..", render it and look at the estimated printing time: 1h16m
  (while printing: with normal 100% it's too slow)
* then open the file "2xwall..", render and look at the estimated printing time : 1h23m

note: *Slicers settings are unchanged*

_Expected Results_
 * in "wall..": from Layer 4: faster print (at 100% feedrate)
 * in "2xwall": it prints as expected

_Actual Results_
 in "wall.."
unexpected slow print speed starting on layer 4.
Estimated printing time should (roughly) be the half

_Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]
[image]

_Is this a new feature request?_
nope. I regard this as a bug

#### Project File (.3MF) where problem occurs
here is the zip [SlowPrintingPerimetersBugFiles.zip](https://github.com/prusa3d/PrusaSlicer/files/4167636/SlowPrintingPerimetersBugFiles.zip)
containing
 - wall pen holder_full.. .3mf
 - wall pen holder_full.. .gcode
 - 2xwall pen holder_full.. .3mf
 - 2xwall pen holder_full.. .gcode

rongith, 2020-03-13:
Your "slow down when layer time is below" value is 20s. You have 219layers then 20sx219=4380s=1h13min. That is the minimal time to print your part (excluding the first layer case that might be bigger than 20s). If you want to decrease the print time, decrease "slow down when layer time is below" or set it to zero or disable "enable auto cooling".

3Dietrich, 2020-03-13:
yea, I thought already about it (and was too lazy to update this bug) 
confirmed. closed. thank you.

closed 2020-03-13 by 3Dietrich