prusa3d/PrusaSlicer #3616: Slic3r PE 2.2.0 Alpha2 and 2.1.1 Lockup

muelli01, 2020-02-07:
### Version
2.2.0 alpha2 linux x64, 2.1.1 linux x64

### Operating system type + version
Ubuntu 18.04.3

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
Slicing the attached thing ( https://www.thingiverse.com/thing:4030633 ) with support (for the tiny letters in the side of the case) will lockup Slic3r at stage "Preparing infill..."

Reproduce: Open Slic3r, load the STL onto printbed, select infill (tested 15% or 20%), enable "Supports everywhere", accept the following dialogue recommendation and start to slice.

* _Expected Results_
Sliced STL

* _Actual Results_
Lockup in stage "Preparing infill..."

#### Project File (.3MF) where problem occurs
Attached.

muelli01, 2020-02-07:
For some reason the attachment disappeared. Tried again.
[G-RX8_housing.zip](https://github.com/prusa3d/PrusaSlicer/files/4169587/G-RX8_housing.zip)

neophyl, 2020-02-07:
I had a look at it.  If you leave it long enough it does complete on my system. 

That small model has nearly quarter of a million facets.  Its basically taking LOTS of processing to slice the really detailed model.
If you simplify the slicing task by setting Print Settings>Advanced>Resolution to something like 0.01 it slices MUCH quicker and as far as I can tell doesn't seem to have lost anything from the model.

muelli01, 2020-02-07:
OK then my system might just be too slow. I waited for 15min or so....
I will close this then. Thank you!

closed 2020-02-07 by muelli01