prusa3d/PrusaSlicer #3617: Sequential printing needs reworking

Fra085, 2020-02-07:
### Version
2.2.0 alpha2 + win64
2.1.1 + win64

### Operating system type + version
Windows 10 Home + 1903

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S MMU2S + 3.8.1

### Behavior
* At the moment sequential printing is unusable because of the continuous error of invalid data
* Auto arrange tool is not able to correctly arrange them so I have to do this manually, trying to fit the models and offsetting them in order to avoid the gantry when printing but usually this doesn't work, even if they are placed at the greatest distance from each other in the bed corners
  * I'm trying to use this function in order to reduce stringing during travels and mainly to take advantage of the MMU unit: it would be great to print every object in a different colour without changing filament every layer but only when it needs to print a different model, wasting a minimum amount of plastic
* If this works, the preview doesn't show any sign of sequential printing. Sometimes to make it works I need to increase the gantry height value over the maximum height of the models
* I should be able to fit the models more easily and control the sequence of the printed objects. Furthermore, MMU compatibility should be better implemented because at the moment the purge tower is not even supported and the slicer is not able to produce a gcode with sequential models  printed in different colours

bubnikv, 2020-02-19:
would you please retest with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha4
please see the release log about the sequential print changes.

Please update your issue with what you believe remains to be fixed or improved. Thank you,

Fra085, 2020-02-19:
There are still some modifications to do for sequential printing:
- Auto arrange tool is still not able to correctly arrange the models because it doesn't offset them in order to avoid collisions and tells "invalid data". If I manually place the models correctly (as shown in the attachment, image-1) it is able to slice them.
- In order to work for taller models (taller than the height of the x-axis gantry, 25mm at the moment) you have to increase this value: if the extrude moves in casual order, the gantry could collide with some models but I'm sure that printing from front to back offsetting the models (image-1) it will work. I know it could be hard to achieve automatically but it would be great. It could be possible maybe programming the travels from one model to another over the height of the tallest one and then landing vertically over the place of the following model.
- The preview still doesn't show any sign of sequential printing.
- The purge tower for MMU2 is still not supported. This is maybe more difficult than the other things but it should be found a way to change extruder (with the MMU2). An idea? Maybe placing a personal purge tower for each model (near or around it)(image-2), as a shell or a large skirt (for example if it happens for a change from model-1 with extruder-1 to model-2 and extruder-2)

image-1
[image]

image-2
[image]

Thanks for your work

Fra085, 2020-02-19:
I have just discovered other bugs in MMU2 mode:
- the wipe options effects are no more displayed in the preview (but they still works since the purge tower has been reduced). In the image-3 you can see this problem knowing that:
model1: default extruder
model2: extruder 2
model3: wipe into the infill
model4: wipe into the infill and into the object
- when the multipart object is completed, the wipe object section higher than that height is skipped (image-4, image-5)

image-3
[image]

image-4
[image]

image-5
[image]

bubnikv, 2020-02-20:
>I have just discovered other bugs in MMU2 mode:

I believe everything is all right here, the confusion raised because the wipe has the same color as the object.

bubnikv, 2020-02-21:
> I have just discovered other bugs in MMU2 mode:

Would you please provide your 3mf to reproduce this issue?
We discovered a regression bug, which we hopefully fixed with 29086aa00bc28a0cccd68fefc11e722b1f023f0d

Fra085, 2020-02-22:
Yes, sure. 
In all of the files, the preview bug of the correct colour infill displaying persists.
The incomplete purge model bug is shown in the first 3mf file.
In the second one, I found out that adding a normal model without any wipe option fixes the previous bug.

3mf files can be found in these link:
https://drive.google.com/open?id=1Nb0FgvQEtte0v6UBRTC80i92x0k_zjTi
https://drive.google.com/open?id=1yhtKqyaC96KAzFiDK3D5rRdmP718sQKY

bubnikv, 2020-02-22:
>In all of the files, the preview bug of the correct colour infill displaying persists.

You mean it persists in the beta?

bubnikv, 2020-02-22:
I retested with PrusaSlicer 2.2.0-beta and the issue seems solved to me. Would you please confirm?

Fra085, 2020-02-22:
I have just downloaded and tested the beta version: all the bugs I have found before are gone.
I'll dig more in this version but I have just found a bug of the print settings window on the right of the plate: the entire right side is out of the screen.
None of the functions I have asked for has been implemented yet.
I'll make you know if I find more bugs.

Fra085, 2020-02-22:
This is the screen that shows the bug.

[image]

lukasmatena, 2020-02-22:
> the entire right side is out of the screen.

This is not nice. It does not show on your screenshots from alpha4 though. Has it started with the beta and shows all the time or do you need some specific steps to get to that state?

> I'll make you know if I find more bugs.

We'll be grateful. Even more if you file a separate issue for each of them. It is very difficult to make sense of it when we're discussing ten bugs and feature requests at once.

Fra085, 2020-02-23:
No problem: I'll open a new discussion for each bug, right?

The right side is out of the screen and shows all the time since I open the program; it has started with the beta version only.

bubnikv, 2020-02-23:
>  No problem: I'll open a new discussion for each bug, right?

yes please.

bubnikv, 2020-02-25:
>This is the screen that shows the bug.

Response by @YuSanka (currently on vacation):

I believe, that “translated” button is a little bit wider, than regular. And I was looked to the code. :( My last changes for options alignment can cause this bug. I think, that adding “&& m_show_modified_btns” to the condition on the 132 line in OptionsGroup.cpp should fix it.

Fra085, 2020-02-25:
You are right: I tried changing languages and seems it happens only for "Italian language" when I use the MMU preset. With single extruder presets it doesn't happen. I didn't check all the languages but for the ones I checked, it doesn't happen.

bubnikv, 2020-03-07:
@YuSanka fixed the right panel layout issue.
I am closing this issue. If you think something of this issue is still relevant (there were apples mixed with oranges in this issue), please open a new one.

closed 2020-03-07 by bubnikv

Fra085, 2020-03-07:
Perfect. Thanks for the fixing.