prusa3d/PrusaSlicer #3619: [Feature Request] 1st Layer "outer thin Wall" Speed same value as upper Layer

El0nMusk, 2020-02-07:
Would be great to leave the value from "outer thin Wall Speed" at same for first Layer. e.c. at 5mm/s 

In my case it moving faster (because 1st Layerspeed at 25mm/s)

So let 1st Layer at the same 5mm/s Speed not 25mm/s specially for Drillholes (Screws etc.) if possible. Most of these Outlines doesnt stick well to the bed, because of fast movements at small diameters (Circle, Holes). Or get bend up to nozzle.

Thx lot

Salamandar, 2020-02-08:
I've stopped using a fixed first layer speed a long time ago.

You can set the speed to a percentage of the other layer speeds (mine is 35%, but why not 10%). Like that you will have different speeds on the first layer.

Also, why not reduce the first layer speed ? It would be a very small print time augmentation.

jesusbackflips, 2020-02-10:
> I've stopped using a fixed first layer speed a long time ago.
>
> You can set the speed to a percentage of the other layer speeds (mine is 35%, but why not 10%). Like that you will have different [...]

jesusbackflips, 2020-02-10:
nice < 3 thank you for respond. i missed that

closed 2020-02-10 by jesusbackflips