prusa3d/PrusaSlicer #3622: Interior faces in a model cause issues with hollowing.

BeldrothTheGold, 2020-02-07:
_build_
version_2.2.0-alpha3-316-gbd7d5cf46

I purchased a model which has hidden interior spaces in it. (News to me prior to trying to hollow it!)
This definitely throws off the hollowing feature a bit. Negative spaces become positive spaces inside the object creating floating objects inside the hollow cavity.

Maybe for the first release it would be best to just warn users when the tool detects negative space inside an object? Similar to how it warns of toolpaths outside the print area or some objects hidden, etc?

In a future release maybe add an option to always create negative space regardless or interior holes in the object?

Model in question showing interior cavities prior to hollowing. 
[image]

After hollowing, shows issue. 
[image]

I can't share the original object, but here's a simple STL with an empty space inside. 
[image]

It shows the same behavior when hollowing.
[image]

Here's the cube within cube mesh. 
[Interior facing mesh.zip](https://github.com/prusa3d/PrusaSlicer/files/4172813/Interior.facing.mesh.zip)

BeldrothTheGold, 2020-02-07:
P.S. 

The normals of the faces within the cube aren't being flipped. Seems this is causing PS to not even attempt to generate supports for the cube within the cube.

bubnikv, 2020-02-09:
Your test cube with a cavity, what is the orientation of the outer / inner surface? Are the normals of the outer surface pointing outside and are the normals of the inner surface pointing inwards? Then such a model would subdivide the 3D space correctly to a cube with a cavity.

If on the other hand the normals of the inner surface were pointing out (that is in the same direction as the outer surface), then the inner surface would be invalid and running the model through netfabb would remove it, which seems to be the case. You may run your commercial model through the netfabb fixing service from the object pop up menu as well, it should help a lot.

We may try to do some tricks, but in general hollowing for such models will be difficult.

BeldrothTheGold, 2020-02-10:
> Your test cube with a cavity, what is the orientation of the outer / inner surface? Are the normals of the outer surface pointing outside and are the normals of the inner surface pointing inwards? [...]

Regarding the test cube. Normals of outer surface are facing out, normals of inner surface facing inward. Created with Blender3d with Boolean Modifier and exported as STL.

For the commercial model, same situation. Outside normals facing out, inside normals facing in. No errors when running it through net fab. I was able to remove the inward facing surfaces using Blender3d to produce a mesh suitable for hollowing. 

I just wanted to make the team aware of the issue.

BeldrothTheGold, 2020-02-11:
Confirmed, the major issue is resolved. The sliced output for both the both the commercial model and example I provided should print successfully. No longer attempting to print unsupported invalid volumes inside the mesh. 

You rock!

bubnikv, 2020-02-11:
@BeldrothTheGold thanks for heads up.
Thanks @tamasmeszaros for solving this issue.
Could you close this issue then?

BeldrothTheGold, 2020-02-11:
I don't mind closing the issue. 

There are still some minor graphical issues related to this, but they won't impact printing. 

[image]

closed 2020-02-11 by BeldrothTheGold