prusa3d/PrusaSlicer #3623: Print individual verticals first of the same object

alyyousuf7, 2020-02-07:
Recently I printed a model which resembled a 4 legged table. Obviously, I had to print it upside down.

The table top printed first and nicely, but when it started printing legs, it faced with two issues.
First being the waste of time due to constant movement from one leg to another and then 3rd and 4th in each layer.
Secondly, the stringing due to constant movement.

What I propose is something similar to "Print Individual parts completely" but for the verticals of the same part.
While printing, I realised that each leg was far enough that if it printed each leg completely and then moved to another leg, it would not collide with the previously printed legs.

This will save power, time and quality.

For this, the slicer needs to know the dimension of the printer head so that it can "group" together the verticals for printing.
After grouping, it must first start printing the vertical groups that are in the front. Because if we start printing from back, they might collide with the head when printing the front later.

foreachthing, 2020-02-07:
I like this idea a lot!
I don't know how much time (or power) I'd save with my Ultimaker 2 ... since the nozzle only sticks 2-3 mm out of the shroud. But it would reduce stringing a lot; only about every 15th layer (@0.2 mm layer height) instead of every single one.

BeldrothTheGold, 2020-02-07:
I think this may be a duplicate to: #3600 Sequential Printing by Height [Feature Request] feature request sequential print

alyyousuf7, 2020-02-07:
Yes, I meant exactly the same thing. Thanks for pointing out!

Closing.

closed 2020-02-07 by alyyousuf7

worldspawn, 2020-10-12:
Humbly suggest you also go there and give it a thumbs up 👍

luigim81, 2021-12-02:
This is the only feature why i am using s3d...