prusa3d/PrusaSlicer #3624: Infill for large heights and nozzle diameters

podarok, 2020-02-07:
Hi.
Thank you for the great software.
I have an issue/feature request
[image]

When it comes to printing large structures(300mm+) with large nozzles (1mm+) and large heights (0.6mm+) seems like there is no good recipe to avoid issue with jumping head(effector in my case - magnetic delta printer)

Example on a screenshot - when using low density infill - every second layer starts to jumping over previous one on cross sections. Because in that spot thickness of the previous layers way higher than between. In small scale this issue not causing problems, because overall mechanics of the printer capable to push through the point without issues.
For that we need to have an infill that will not have crossed trails, just points touching neibours on a side, not on top/bottom.

Could you recommend anything?

Making infill thinner helps, but it becomes not usable in terms of structural rigidity and contains a lot of holes right after jumps.

Thank you in advance

BeldrothTheGold, 2020-02-07:
PrusaSlicer supports several infill patterns which avoid crossing crossing extrusions on the same layer. 
Some examples: Rectilinear, Gyroid, 3D Honeycomb, Hilbert Curve, etc.

These options are all available under the Print settings tab. 

[image]

podarok, 2020-02-08:
Thank you. Seems like slow speed(15%), 0.65mm layer height, 165% width, 1mm nozzle and honeycomb - does work
[image]

closed 2020-02-08 by podarok