prusa3d/PrusaSlicer #3626: Wipe tower support for absolute extruder addressing printers

Highlanderes, 2020-02-08:
PrusaSlicer 2.2.0_alpha2

For dual extruder 3D printers running absolute addressing, specifically TAZ6 (with FlexyDually V2 tool head) running Marlin in a dual extruder multi material print.

Enabling wipe tower for a dual extruder printer running absolute addressing, results in an error message preventing slicing: 

[image]

With further support for 3rd party printers planned for 2.2.0, I hope to see WipeTower for absolute extruder addressing. It is needed for succesful Dual extruder printing to prime extruders especially with oozing prone materials such as Ninjaflex.

To provoke this, open the attached file and open print settings, multiple extruders, enable prime tower and the error message pops up.

Note: with PrusaSlicer 2.2.0 - alpha2 I can not actually see the specific error message content on my macOS 10.12.6 (16G2016) (another issue), but I can se it opening with PrusaSlicer 2.1.0

[TAZ6 FDv2 Factory Printer test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4174511/TAZ6.FDv2.Factory.Printer.test.3mf.zip)

#718 includes a way for users to build their own prime towers manually, but may be limited to one multi extruder multi material object.

#2198 suggests multi extruder multi material wipe tower is already in place, but the error messages here indicates this is a new feature request.

Highlanderes, 2020-02-27:
Unfortunately the #718 work-around does not work, as the dual model is always printed before the towers regardless of sequence loaded. So this is a blocking issue at this time.

photodude, 2020-02-29:
Can confirm with a BIBO2 dual extruder. When set to absolute extrusion mode the wipe tower is not available and an error comes up saying the extruder must be in relative extrusion mode to use the wipe tower.

+1 for adding wipe tower functionality to the absolute extrusion mode

DC5TransportR, 2020-12-17:
I am using PrusaSlicer with my Geeetech A30M and would love to have the prime tower feature enabled for absolute extrusion not only relative. The mixing nozzle is 2 in 1 out and would definitely love to be able to prime virtual extruders properly with a tower.

eMPai-git, 2021-04-27:
I want to use PrusaSlicer with a FlashForge Creator Pro 2. IDEX works but unfortunately the Creator Pro 2 doesn't support M83 (yet). So no Wipe Tower for me.

The workaround **does** work for me by adding the pillars **last**, e.g.:

- Dualcolor-Object-1 - Color 1
- Dualcolor-Object-2 - Color 2
- Shape-Box - Color 1
- Shape-Cox - Color 2

closed 2026-09-29 by prusaslicer-bot[bot]