prusa3d/PrusaSlicer #3627: implement 'Wipe to Interface' for MMU - Feature Request

ReactSub, 2020-02-09:
### Version
Feature request. Regards https://github.com/prusa3d/PrusaSlicer/issues/2671 and https://github.com/prusa3d/PrusaSlicer/issues/3390

and lays the groundwork for the most efficient implementation of:
https://github.com/prusa3d/PrusaSlicer/issues/3397 **multiple support material (MSM) - Feature Request** 

### Operating system type + version
N/a

### 3D printer brand / version + firmware version (if known)
Prusa 3MKS w/ MMU 2

### Behavior
* _Describe the problem_
Soluble Interface feature transitions from spool to spool purely via the purge tower. 

* _Steps needed to reproduce the problem_
Print with Soluble Interfaces enabled. 

* _If this is a command-line slicing issue, include the options used_
N/a

* _Expected Results_
Amalgamate "Wipe to infill" and "Wipe to object" into a feature as "Wipe to Soluble Interface"
By wiping from the primary support material to soluble as part of the throw-away block (where there is enough material used to do so effectively), we reduce the use of the purge tower, further increasing the economy of the "Soluble Interface" feature. 

* _Actual Results_

* _Screenshots from __*PrusaSlicer*__ preview are preferred_
N/a

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs

bubnikv, 2020-02-09:
> By wiping from the primary support material to soluble as part of the
throw-away block (where there is enough material used to do so
effectively), we reduce the use of the purge tower, further increasing the
economy of the "Soluble Interface" feature.

It does not make sense to me. You want to have the soluble interface clean,
otherwise the non-soluble whiskers will stick to your object and you will
have to remove them manually. It defeats the purpose of soluble interfaces.

ReactSub, 2020-02-09:
Ok, but it makes sense to me. And you quote the important bit "where there is enough material used to do so effectively". Let my help you out. I think we're getting hung up on the idea that the lead up has to be contiguous, which isn't the case.

In fact, I'm proposing the opposite of what you're citing as a potential issue. To clarify, there is no reason that soluble support has to be _only_ touching the actual model. It can be used as part of the bulking agent of the throwaway solid support. If that solid support is riddled with tiny holes after the dissolving post-process step, it doesn't matter since the whole thing is going in the trash anyway. This reduces size of the needed tower by (potentially, based on the model) begetting less wasted support and soluble material.

I'll draw you a picture.

[image]

So this is a cross section of a wingtip and second, separate overhang being printed. The outline of the wing and overhang are a projection of the final model, since you can see we're a few layers short of completion. In each case of this 2D representation, the left side of the layer prints first, then the right. For the bottom four layers, there is enough leadup on just the right side to transition completely to the soluble material before starting the model. The top four layers do not have enough leadup on the right to transition to soluble, so the wipe starts on the left. 

If the second, random overhang was not there, we would likely not have the lefthand support layers, so we would be more reliant on purge block, since there would not be enough combined leadup in the layer.

So, again, the green bits on the left edge of the left side of the top four layers will dissolve, but it's irrelevant since the green still functions as support during the print.

closed 2026-09-29 by prusaslicer-bot[bot]