prusa3d/PrusaSlicer #3630: Prusaslicer keeps crashing and disappears from desktop

deafoldgit, 2020-02-09:
### Version 2.1.1
_Version of PrusaSlicer used goes here_2.1.1

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_ I tell the slicer to slice, it starts the prusaslicer just disappears of my monitor screen, so have to re start slicer
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-02-09:
What is your system? Did any of the older PrusaSlicer versions work on your computer? Does the last alpha work for you https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha4 ?
Do you have your graphics card drivers updated? What is your graphics card?
Do you have any 3rd party application installed, which may hook into other applications?
What happens exactly?

deafoldgit, 2020-02-09:
Hi, windows 10 32 gig ram 16 gig video ram 4 tb of hard drive. Ted

deafoldgit, 2020-02-09:
Hi, all drives are up dated, video is g force 10, Alpha did work fine for me, no other software to make it crash as far as I am aware. Nothing else running at the same time apart from internt browser (Opera). Ted

bubnikv, 2020-02-09:
cool, so the issue is solved for you with the PrusaSlicer 2.2.0-alpha4?

deafoldgit, 2020-02-09:
Great Thank you, Ahoj

bubnikv, 2020-02-09:
thanks. Closing.

closed 2020-02-09 by bubnikv