prusa3d/PrusaSlicer #3631: PrusaSlicer-2.2.0 alpha4 linux can't run

killerb77, 2020-02-09:
### Version
PrusaSlicer-2.2.0-alpha4+linux-x64-202002091020

### Operating system type + version
Pop OS 19.10

### Behavior
When trying to run I get this error message: error while loading shared libraries: libmpfr.so.4: cannot open shared object file: No such file or directory

Also, when I try to run the appimage version nothing opens at all.

Alpha 2 runs fine.  Problem started with Alpha 4.

Area5142, 2020-02-09:
OS: Linux Mint 19.3 64bit
PS: PrusaSlicer-2.2.0-alpha4+linux-x64-202002091020.AppImage

Have same problem, missing library. Output on console:
```
~/bin/AppImages$ ./PrusaSlicer-2.2.0-alpha4+linux-x64-202002091020.AppImage
/tmp/.mount_PrusaS40la9O/usr/bin/bin/prusa-slicer: error while loading shared libraries: libmpfr.so.4: cannot open shared object file: No such file or directory
```
Alpha 2 works fine here too.

bubnikv, 2020-02-09:
Alpha4 requires the gmp and mpfr libraries for exact arithmetics, which is
used by the SLA hole drilling algorithm. I suppose we are linking the
libgmp and libmpfr dynamically by mistake.

```
bubnikv@bubnikv-VirtualBox:/tmp/.mount_PrusaSOouLbP/usr/bin/bin$ ldd
prusa-slicer | grep gmp
libgmp.so.10 => /usr/lib/x86_64-linux-gnu/libgmp.so.10 (0x00007f59604a8000)
bubnikv@bubnikv-VirtualBox:/tmp/.mount_PrusaSOouLbP/usr/bin/bin$ ldd
prusa-slicer | grep mpfr
libmpfr.so.4 => /usr/lib/x86_64-linux-gnu/libmpfr.so.4 (0x00007fdf8b2d4000)
```
You will solve the issue for yourself if you install the two libraries from
the distro. We will discuss internally whether we will bundle the two
libraries with the AppImage, or whether we will link statically.

BerndJM, 2020-02-09:
OS: Linux Mint 19.1 64bit
PS: PrusaSlicer-2.2.0-alpha4+linux-x64-202002091020.AppImage

May be a "first aid":
There is a `libmpfr.so.6.0.1` already installed (with a symbolic link `libmpfr.so.6` pointing to ist).
Just making a symbolic link `libmpfr.so.4` pointing to the `libmpfr.so.6.0.1` works fine (at least for me).

Area5142, 2020-02-09:
I tried to (re)install libmpfr.so.4 (MPFR 3.x), but it seems not longer supported from Ubuntu 18.04 (Linux Mint is based on Ubuntu). Source of information: https://stackoverflow.com/questions/51288467/cannot-open-libmpfr-so-4-after-update-on-ubuntu-18-04

From the same source is a hack to make the old version of the library work, using symbolic link:
`sudo ln -s /usr/lib/x86_64-linux-gnu/libmpfr.so.6 /usr/lib/x86_64-linux-gnu/libmpfr.so.4`

This hack makes a link for the missing old file to the newer supported libmpfr.so.6.

I tried the hack and it works - alpha4 now starts and displays the first time wizard. Haven't discovered any side effects yet.

killerb77, 2020-02-09:
My system also has libmpfr.so.6 installed.

I downloaded and installed libmfr4 from https://packages.ubuntu.com/xenial/libmpfr4.
I downloaded the amd64 .deb file as my system has 64 bit pop os installed (which is ubuntu based) and did sudo dpkg -i libmpfr4_3.1.4-1_amd64.deb.

It installed and now I can run the appimage just fine.

rene-win, 2020-02-09:
sudo ln -s /usr/lib/x86_64-linux-gnu/libmpfr.so.6 /usr/lib/x86_64-linux-gnu/libmpfr.so.4

You probably have a newer version of the lib, just symlink it to version 4 and it works.
At least it solved the problem for me on ubuntu 18.04

a-allard, 2020-02-09:
I also am having this issue on both my Ubuntu machines.

Mc261, 2020-02-09:
> I tried to (re)install libmpfr.so.4 (MPFR 3.x), but it seems not longer supported from Ubuntu 18.04 (Linux Mint is based on Ubuntu). Source of information: [...]

Exactly that helps me strating Prusa-slicer 2.2 Alpha 4! Thanks for it!

Pelado-Mat, 2020-02-10:
Same problem here:

> m@carbon:~/bin$ lsb_release -a
> No LSB modules are available.
> Distributor ID:	Ubuntu
> Description:	Ubuntu 19.10
> Release:	19.10
>Codename:	eoan

Workaround using @Area5142 instructions

> I tried to (re)install libmpfr.so.4 (MPFR 3.x), but it seems not longer supported from Ubuntu 18.04 (Linux Mint is based on Ubuntu). Source of information: [...]

bubnikv, 2020-02-10:
I have updated the Linux x64 buils on
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha4
with version, which links the two mentioned libraries statically. The new builds have a timestamp of today afternoon.

Thanks for heads up. Closing.

closed 2020-02-10 by bubnikv