prusa3d/PrusaSlicer #3632: [BUG] PrusaSlicer 2.2.0 alpha 4 on macOs - Application crash with some STL

BastR, 2020-02-09:
### Version
PrusaSlicer 2.2.0 alpha4

### Operating system type + version
macOS X Catalina 10.15.2

 
### 3D printer brand / version
Homemade printer with marlin 2.0 bugfix

### Behavior
Fist, PrusaSlicer is slower than alpha2, when I try to import an STL, the app freezes for a few seconds whereas it was instantaneous in the previous version.

If I open, for exemple, Hollowcube.Stl there is no problem, but if I open an stl that I use for calibration then the application always crashes. I join macOS crash log and the STL that crashes the application.

[Crash dump OSX.txt](https://github.com/prusa3d/PrusaSlicer/files/4177168/Crash.dump.OSX.txt)

[20x20x15_calibration-shape.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4177170/20x20x15_calibration-shape.stl.zip)

bubnikv, 2020-02-09:
Thanks for the report. I can reproduce the crash on Windows with this
particular model as well.

bubnikv, 2020-02-09:
The crash is likely new. You likely forgot to enable thin walls, therefore no extrusion is generated and due to some refactoring of the G-code generator code it crashes now.

I have just added a check 519a982fe971910b89f459229873f7bb8cac4b63 to show an error message to the user instead of crashing. It will be inside the next alpha or hopefully beta.

[image]

> PrusaSlicer is slower than alpha2, when I try to import an STL, the app freezes for a few seconds whereas it was instantaneous in the previous version.

That is another issue, we have to investigate the claim.

BastR, 2020-02-09:
That's exactly right, if I activate the thin wall detection beforehand, the application works correctly.

Do I need to open an other issues for the app freezing when importing STL ?

bubnikv, 2020-02-09:
>  Do I need to open an other issues for the app freezing when importing
STL ?

I don't think so. If some STLs are worse than the others, please provide
them here.

BastR, 2020-02-11:
I made more tests today and I don't know, but the slowdowns I had on the first few launches are gone.

bubnikv, 2020-02-22:
The crash will be fixed in toda's beta. Closing.

closed 2020-02-22 by bubnikv