prusa3d/PrusaSlicer #3633: PrusaSlicer frequently leaves thin walls hollow (regardless of Detect Thin Walls setting)

mjparme, 2020-02-09:
### Version
[image]

### Operating system type + version
MacOS 10.15.3

### Behavior

If an object has thin walls the wall are frequently left hollow. As an example I have included a 3MF file for an object I am having this issue on. I have also posted a screenshot of PrusaSlicer preview and also Cura preview. Cura slices it correctly. This isn't the only model I have experienced this on. Usually when I run into this problem I just switch over to Cura to slice the models PS does this on. 

PrusaSlicer: 

[image]

Cura:

[image]

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs

[PegboardScrewdriverHolder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4177281/PegboardScrewdriverHolder.3mf.zip)

the78mole, 2020-05-08:
Same here. Instead of filling it with reduced extrusion width, PrusaSlicer keeps it hollow, regardless of the "Detect thin walls" setting.

[image]

neophyl, 2020-05-08:
Detect thin walls is used to print walls were the thickness would normally be too thin for a single extrusion.  Your walls are thick enough so the thin walls isnt a factor.
Do you have gap fill enabled ? (gap fill speed must be non zero)

the78mole, 2020-05-11:
Ok, gap fill was the trick that helped... Thanks

closed 2020-05-20 by mjparme