prusa3d/PrusaSlicer #3634: Brim and loops make first layer crease

acorchia, 2020-02-09:
### Version
2.2.0 apha2 and apha4

### Operating system type + version
_Win10 64

### 3D printer brand / version + firmware version (if known)
MK3S 3.8.1

### Behavior
In Print setting activate loops =1 by default
Then on aspecific print cicl on enable brims.
First layer is crease : Loops is in the middle of brims

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

[Mon Mousquetons.zip](https://github.com/prusa3d/PrusaSlicer/files/4177357/Mon.Mousquetons.zip)

May be activate Brims shoudl disable loops ?

BeatSlayer, 2020-02-09:
This was a requested feature added to slicer in 2.2.0 alpha 1.

From release notes:

*The skirt is newly allowed to overlap with the brim, so that the skirt, when used as a wind draft shield is stabilized by the brim. Brim lines are split at the intersections so there is no overextrusion [#724](https://github.com/prusa3d/PrusaSlicer/issues/724)

tilllt, 2020-02-14:
There is a seriously weird behaviour in the Alpha 4 Brim creation. I am trying to print this battery organizer and the brim alone takes like 30 min to print because of totally erratic travel movements (which cause stringing etc).

[AAA_Organizer_37Cell.STL.zip](https://github.com/prusa3d/PrusaSlicer/files/4204285/AAA_Organizer_37Cell.STL.zip)

lukasmatena, 2020-02-17:
@acorchia 
@BeatSlayer is correct. This was indeed requested and it makes sense, as discussed in #724.

However, what @tilllt reportes is not correct (erratic order of the split extrusions). I hope I fixed it with 9a84422. It should be ok in upcoming 2.2.0-beta. Thanks for the report.

Closing.

closed 2020-02-17 by lukasmatena