prusa3d/PrusaSlicer #3635: [Feature Request] Add more options to the Height range modifier

th33xitus, 2020-02-09:
### Suggestion
At the moment there are some quite nice options available to choose from in the height range modifier.
But some additional parameters could be quite handy or usefull imho.

**Layers and perimeters:**
- Layer height
- Ensure vertical shell thickness
- Avoid crossing perimeters
- Detect thin walls
- Seam position
- External perimeters first

**Infill:** 
- Fill angle
- Infill before perimeters 

**Advanced:**
All Extrusion width related entrys should be listed under the Advanced Settings like in the "Print Settings" tab. At the moment there are some extrusion width settings within an own category.
- Default extrusion width
- Perimeters
- External Perimeters
- Infill
- Solid Infill
- Top solid infill

**Filament:**
- Extrusion multiplier
- Extruder temperature

**Cooling:** (maybe)
- Fan speed

**Extruder:**
- Retraction length
- Lift Z

I know it is quite a lot but it would make the height range modifier an extremely powerfull tool. 
You could actually almost print two objects on the same build plate with a complete different set of settings in one go. It could be (at least kind of) compared to Simplify 3D's so called processes you can set up, which let you choose from the full spectrum of settings S3D has to offer and apply them to different parts of your print (depending on z, so exactly like the height range modifier here) or assign those processes to different objects to your build plate if you have more than one object on there and go for sequential printing for example.

If there are any question why i think option x makes sense to be included there, just ask and i'll try to explain with an example.

Doprintityourself, 2020-12-15:
Please add more heigt range modifier (cooling)

I have massiv problem printing ABS, with falling in vertical walls for the first layers. So I would need "cooling for the first layers" and than switch it off with a height range modifier

Thank you very much in advance!

mgx0, 2020-12-19:
Please add modifier for Cooling/Fan Speed. This is how it could look like:

putting a modifier object in place:
[image]

and selecting type of modifier > Advanced > Fan Speed or whatever
[image]

newphreak, 2021-04-03:
Can we get this going in the next version? This would bring the slicer to the next level!

MrKumaPants, 2021-04-03:
SuperSlicer has this and it's a fork of PrusaSlicer. The ticket is almost a year old. Would be really nice for calibration tests.

fionnb, 2022-03-22:
+1 for cooling fan modifier! This is really essential for some complex prints.

Kekht, 2022-03-24:
+1 for cooling fan modifier! This is really essential for some complex prints.

WindoC, 2022-03-25:
+1 for cooling fan modifier! This is really essential for some complex prints.

Dephrilibrium, 2022-05-07:
Would also be pretty interested in **retraction length** for a certain amount of layers, due to I get pretty bad results in the yellow region because the travel distance is pretty high and some filament leaks from the nozzle.

[image]

fionnb, 2022-05-08:
> Would also be pretty interested in **retraction length** for a certain amount of layers, due to I get pretty bad results in the yellow region because the travel distance is pretty high

I get your point, but man, why don't you print this sideways? Less layers = more stability. Besides it should print quicker und with less need for support.

Dephrilibrium, 2022-05-08:
On one hand it's to big for all other positions, and on the other hand it has not parallel side-walls (as you expected). It's designed with a small angle like a "v" from a relatively far point. Therefore i would get pretty big and ugly areas on the top layer from the slope. Furthermore it has channels along it's complete back. Then I change the current support with support underneath the part and within a very long and small notch and I don't want to remove that.
[image]
[image]
[image]
(for the last screenshot I scaled the model down a bit to be able to slice it for the showcase here)

But indipendend from my application and part, that feature would be a cool feature anyway.

dartrax, 2022-07-19:
+1 for Lift Z !

Kekht, 2022-08-31:
+1 for Fan speed

Plohish67, 2022-12-10:
+1 for Fan speed

johnchen40904, 2023-01-26:
+1 for Extrusion Multiplier
Trying to create some DefeXtiles following that 2020 MIT paper, EM options with modifier meshes could make some pretty cool stuff like laces.

adamfilip, 2023-02-20:
Please add ability to add Extra length to restart and retraction length / speed as height modifier

mfettig, 2023-06-07:
I would also like to see Lift Z available as a height modifier. In my case I would like to avoid having the nozzle scrape across an ironed surface. This could be a solution for #2118 as well

ChaosBlades, 2023-09-27:
I was looking to change extruder temperature by way of a height range modifier. 

I have 4 HUGE layers at the start of the print and the extruder can't keep the filament hot enough to stick to the previous layer. Either I need to print them way slower or bump the print temp the first 4 layers. If I increase extruder temp through the entire print then I get oozing on long travel moves.

Would like to see this added.

Kekht, 2023-10-02:
> I was looking to change extruder temperature by way of a height range modifier.
>
> I have 4 HUGE layers at the start of the print and the extruder can't keep the filament hot enough to stick to the [...]

This could be done easyly bu using scripting and "Custom g-code before layer change"
```
{if layer_num == 5}
M104 S{first_layer_temperature[0]-10}
{endif}
```

But if you have this problem - probably you are trying to print faster, than that polymer could be printed on your hotend.

MooMinIL, 2024-01-29:
Glad I found this open issue:
I second the need for all 'Filament Overrides' settings to be available as modifiers.
This is critical for making quick and easy temperature/retraction/stringing towers to calibrate new filament profiles while using specific printer profiles (with/without input shaper) and print settings.

Right now for example I want to test out if a stringy filament can be mitigated by using Travel Lift changes, Retraction speed changes or the new Travel Lift Ramping.

I was planning on running one g-code with a single test for a combination of these settings but it seems I have to create a separate print file for each settings change I want to test which is extremely time consuming and wasteful when I need to test several new material spools.

Really hope this gets implemented soon!

matej-marjanovic, 2024-03-31:
Lots of great suggestions in this issue. Would just like to second the suggestion for adding the **"Avoid crossing perimeters" option and parameters to be available within the height range modifier.**

I am inserting a previously printed component into a component I am currently printing mid-print and up until that point and a few millimeters after that point I do NOT need to be careful about the print head crossing perimeters but just for those few millimeters it is crucial.

echo-lalia, 2024-05-30:
Just voicing my support for this feature request :) Specifically, fan speed control on modifier objects would be a huge help. 

I'm currently printing an object that uses grid infill as a mesh screen, and it requires some easy bridging on that mesh.   
PrusaSlicer does not detect the infill as a bridge and so it fails. I could change the speed on the modifier, but no way to change the fan control. I might end up having to redesign my part around this limitation.

Big-Channie, 2024-09-19:
Bumping this, being unable to make custom calibration towers is very frustrating. I'd love to stick with Prusaslicer but any time I need to calibrate, I try using the options available and eventually just switch to superslicer or orca. Prusaslicer could just be so much better with user experience features like this.

carlosjln, 2024-11-13:
The Extrusion multiplier option would be very handy, whenever I need this feature unavoidably I have to switch to Orca Slicer for those jobs.

Kekht, 2024-11-15:
> Bumping this, being unable to make custom calibration towers is very frustrating. I'd love to stick with Prusaslicer but any time I need to calibrate, I try using the options available and [...]

Actually for calibration towers (Layer-based changes of any parameter, changeable with g-code) you could use some scripting, here are [some exanples](https://github.com/Kekht/PrusaSlicerGcodeScripts)

stillpointx, 2025-07-15:
I wish prusaslicer had an option to adjust fan cooling  speed for the last 7 top layers and to also change the External perimeters speed in mm/s for the last 7 top layers.   When I tried I got an error saying the spiral vase option can only be used when printing single material objects.

Tupson444, 2026-09-13:
still wanted

fionnb, 2026-09-14:
Definitely still very much wanted!