prusa3d/PrusaSlicer #3636: IDEX extra extrusion before new tool starts printing

botlawson, 2020-02-09:
### Version
2.1.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Hictop D3 Hero IDEX.  Runs an old version of Marlin 1.0

### Behavior
When generating dual material prints, an extra "G1 E-0.30000" is inserted between the Tn and move to start printing.  This causes creates a loose string of material that gets stuck in/on parts.  All un-retracts after tool change should happen over the part at the start of the new tool's tool-path.  Thus the worst they can do is create a small blob.  Older versions got this right.  I can't find where this setting is configured.

G-code excerpt:
`G1 F2400
G1 X131.800 Y155.798 E4.02883
G1 F7200
G1 X132.100 Y155.798 E3.93383
G1 E3.92883 F2400.00000
G92 E0
; Filament-specific end gcode 
;END gcode for filament
;before tool change
M107
T0
G92 E0
; Filament gcode
;end filament gcode
G1 E-0.30000 ;<<<< this should not be here !!!
G92 E0
G1 X146.959 Y150.774 F9000.000
G1 E0.40000 F2400.00000
G1 F2210.87
G1 X153.104 Y150.893 E0.60804
G1 X153.211 Y149.631 E0.65093
G1 X153.466 Y149.635 E0.65958
G1 X153.387 Y150.899 E0.70242`

#### Project File (.3MF) where problem occurs
[lucky_cat_fill_error.zip](https://github.com/prusa3d/PrusaSlicer/files/4178071/lucky_cat_fill_error.zip)

rtyr, 2021-02-05:
I am not sure I understand, why is this a problem? The "G1 E-0.3" is not extra extrusion, it is a retraction (your retraction is 0.4mm but you have also 0.1mm toolchange retraction set, so 0.3mm is rest of the lenght to full retraction distance). Then there is classic full unretract 0.4mm before printing.

rtyr, 2021-03-22:
No response. Closing.

closed 2021-03-22 by rtyr