prusa3d/PrusaSlicer #3637: Object disappears when wipe to object turned on

Diterex, 2020-02-10:
### Version
PrusaSlicer 2.2.0-alpha4

### Operating system type + version
Windows 10.0.183363 Build 18363

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S FIRMWARE 3.8.1 (1.0.6)

### Behavior
When trying to slice with alpha4, if wipe to object is enabled then the object disappears from the build plate. Screenshots have been included. 

[MMU2S issue alpha4.zip](https://github.com/prusa3d/PrusaSlicer/files/4178915/MMU2S.issue.alpha4.zip)

bubnikv, 2020-02-10:
Thanks for the report. It is a regression in regard to 2.1.1.

The "wipe into object" feature is broken. If there is no wipe at a given layer, the layer is not printed at all in 2.2.0-alpha4.

bubnikv, 2020-02-10:
Fixed with 53bfb6bed32f79c5f29b5ce1491e74f9c322e7f1

This bug is a regression of an optimization 3e0690b37bff4af00015a3b393c39a129ab0d1cc

With the optimization disabled, the G-code generator will not be slower
than PrusaSlicer 2.1.1, we can optimize the code in PrusaSlicer 2.3.

The fix will be part of the next alpha (or better, hopefully the next beta).
Thanks for heads up. 
Closing.

closed 2020-02-10 by bubnikv

bubnikv, 2020-02-22:
The optimization was reverted & fixed with https://github.com/prusa3d/PrusaSlicer/commit/53bfb6bed32f79c5f29b5ce1491e74f9c322e7f1

One more bug due to the optimization was fixed with 29086aa00bc28a0cccd68fefc11e722b1f023f0d
(Due to refactoring of G-code export some "don't care" extrusions were not extruded at all.)