prusa3d/PrusaSlicer #3639: Vase mode cannot deal with thin walls

smurfix, 2020-02-10:
### Version
master, 76433da

### Behavior
* Vase Mode leaves holes when a wall becomes "too thin".
* To reproduce:
  * Load Cisar's Vase
  * either reduce to 50% or set the nozzle to 0.8mm
  * scroll down a bit
* _Expected Results_
  * walls are auto-thickened to extruder width so that vase mode still works
* _Actual Results_
  * The slicer leaves holes which aren't very useful when you want an actual vase (water leaks)

[image]

bubnikv, 2020-02-10:
You are expecting too much from the poor slicer. You want it to do design decisions for you.

smurfix, 2020-02-11:
I tend to disagree. I want to be able to print a large vase and a small vase by simply scaling the model (or a more sturdy vase by using a larger nozzle), while having the slicer stay as close to the model's design as possible.

The choice is between the slicer saying "oops, that wall is too thin, so I'll skip it" and "oops, that wall is too thin, but I'll print it anyway". Arguably they're both design decisions; both "work", for some definition of that word, when printing normally. In Spiral Vase mode, of course, the latter would still be OK while the former does not.

bubnikv, 2020-03-15:
There is not an upper limit to the possible complexity of the slicer.

pmottram, 2020-08-25:
How about an Expert check box that follows the (inset) external perimeter of the mesh and does not remove any self-intersecting geometry (material clash leading to a failed print is the users fault hence 'Expert'). Not sure if there are edge cases where this behavior is actually necessary (e.g. at the tip of a pyramid or sharp acute angels).

closed 2026-09-29 by prusaslicer-bot[bot]