prusa3d/PrusaSlicer #3640: Suggestion: eject button after exporting G-Code

gwildavies, 2020-02-10:
Feature request for PrusaSlicer Windows:
I always 'eject' my pen drive or SD card after exporting my Gcode, else Windows gives me a ticking off.
To eject, I have to open Windows explorer, choose the drive to eject, right click, and eject it. It just isn't smooth.
Could you create a button in PrusaSlicer to 'eject' the removable drive after GCode is generated and exported? thanks!

bubnikv, 2020-02-10:
Please try the
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha4
It implements the manual eject after G-code export.

bubnikv, 2020-03-15:
implemented in PrusaSlicer 2.2. Closing.

closed 2020-03-15 by bubnikv

gwildavies, 2020-03-28:
Sorry I don't know what alpha builds are but I've tried in 2.2 and it's great.
And... could you make it so it's one click to export and eject if successful export? Else we have to export, then have to press the new button. Even smoother if we don't need to do both!