prusa3d/PrusaSlicer #3641: Hollowing files resets the view direction

kenshinvt, 2020-02-10:
### Version
_Version of PrusaSlicer used goes here_
2.2.0-alpha4+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
n/a

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
In the hollowing Gizmo clicking "preview hollowed and drilled model" resets the directions of the clipping viewing slider even after a user has changed the direction to a different one.  Please also consider having the direction clipping viewer default to the Z-axis, as this is the most important clipping view in SLA printing (as this shows overhangs/islands).  Preferably with the direction set to the top, since this shows overhangs the most clearly--building the model from the ground up layer by layer.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
Multiple STL files tested.

BeldrothTheGold, 2020-02-11:
I agree the clipping plane work flow definitely doesn't feel intuitive at first. 
It may indeed be best to default clipping to the Z-axis like normal slicing. I do appreciate the ability to align the clipping plane to my current view though. Might not be a bad idea even to add a separate clipping gizmo you can enable and disable separate of the hollowing and support editing.  

Another hollowing related workflow improvement would be a preview of the drilled hole when mousing over the object before clicking. (this would also be nice for support editing)

Also, it would be nice if there  was a tool tip to the reset direction button which says "Align clipping plane with camera view" or something similar to clarify what it does.

Also, I know this isn't exactly related to this issue, but the tooltip when hovering over Hollowing closing distance slider is empty.  (didn't want to open another issue)

[image]

lukasmatena, 2020-03-15:
Please retest in 2.2.0-rc3. The logic was changed so that hollowing does not move the clipping plane if the user already moved it himself.

bubnikv, 2020-03-15:
closing for now. Please reopen if you think it is not working the way it should.

kenshinvt, 2020-03-15:
I just tested RC4.0.  The issue is still persisting (partially fixed), e.g., click on support for an object.  Change the view from the default.  Then click on a different object.  Then click back to the object with the same view.  The view has now been reset to a default.  The issue above with hollowing has been fixed, but the user's direction view change is still getting changed.  Please let me know if this should be a new issue submission as I don't think I can reopen.  Thank you for fixing the original use case.  Please also consider a global view default configuration setting for users (e.g., "based on model" vs "always top down" etc.).

closed 2026-09-29 by prusaslicer-bot[bot]