prusa3d/PrusaSlicer #3642: [Bug] G-Code Export button visible in "non-default" position.

Jebtrix, 2020-02-10:
### Version
Latest git master

### Operating system type + version
Win7 PRO 64bit

### Behavior
When first opening program G-Code Export button is oddly visible below Slice Now button. Once you slice an object visibility of the button goes back to normal. 

[image]

kocikdav, 2020-02-11:
Would you please try if this happens in our build 2.2.0-alpha4?

Jebtrix, 2020-02-11:
Still occurring with latest @ 2e62f03ba88fd3f4908c687efd6f814233a71cee

Jebtrix, 2020-02-11:
Removing profile settings folder and starting from scratch fixes it. The root of the problem is I set a default octoprint_host and octoprint_apikey in Vendor\PrusaResearch.ini to workaround the limitation of PrusaSlicer not saving that information in saved projects. It allows me to quickly use reset default to enter that information when needed. 

I'll leave it up to you if this issue should be closed or handled with a fix.

closed 2020-02-20 by Jebtrix