prusa3d/PrusaSlicer #3646: Recent projects load does not mesh well with NAS or drives that spin down

ruairidh37, 2020-02-11:
### Version
2.1.1 +win64

### Operating system type + version
Windows 10 Pro version 1903 build 18362.592

### 3D printer brand / version + firmware version (if known)
Prusa SL1, Stock, Firmware 1.3.0
Prusa i3 MK3S, Stock (kit), Firmware 3.8.1

### Behavior
Recent projects that are stored on a NAS (or disk) which spins down to preserve power will pop up an error message saying the project is "no available" or similar bad English. This will remove the project from the list, even though the file is there (later.
* _Steps needed to reproduce the problem_
Wait for drive to spin down, try and open the project
*  _Expected Results_
The program will notice the drive is spinning up and wait. The file should only be removed from the list if it is really not there, i.e. does not appear in a listing of the containing folder
* _Actual Results_
Error message with incorrect English, project is no longer in the list of recent projects

_Is this a new feature request?_
No

bubnikv, 2020-02-18:
>  will pop up an error message saying the project is "no available" or
similar bad English

It would help to cite the "bad english" message in its full form, so we can
fix it.

I am not sure whether we can detect your situation? How should the poor
slicer recognize the situation, that your NAS is spinning up the hard
drives and before spinning them up, it returns bogus data?

ruairidh37, 2020-02-18:
I will recreate the issue and grab the exact message.
Other software manages to trigger HDD spin-up and wait before deciding the file does not exist. Note that I get the same with internal disks in my machine that are spun down after long periods of no access. I.e. if I open a recent word file, my disk spins up and the document opens. If I do the same in the slicer, a message pops and the file is removed from the recent files list. If it is somehow impossible to handle the spin-up situation, at least don't remove the file from the recent list or give me an option in preferences that allows to handle the situation?

bubnikv, 2020-02-18:
I don't know. Likely it would be a platform specific implementation. We are
trying to use multi-platform libraries as much as possible to keep the
implementation simple. Also we would have to get hand of a NAS similar to
yours to implement and validate.

ruairidh37, 2020-02-23:
[image]

Here is a screenshot of the error message, "The selected project is no longer available" would be better. The same issue exists with internal drives that spin down after inactivity, that should be easier to replicate.

BeldrothTheGold, 2020-02-27:
Hello @bubnik. This has affected me as well.  Specifically it happens when my NAS didn't reconnect correctly on boot. Not sure why it happens since it's on the local network but it does. I agree it can be frustrating. 

What if, when this happens, the dialog lets the user choose if they want to remove the project from the list? 

It could say something like: 

"Unable to open the selected project. Path was unavailable. Do you want to remove it from the recent projects list?"

Then the user could choose to remove it or do what they need to reconnect to their network drive or what have you.

Something like that?

BeldrothTheGold, 2020-03-05:
You guys rock! @ruairidh37 are you happy with this solution?

ruairidh37, 2020-03-06:
Definitely! Thanks!

closed 2020-03-06 by ruairidh37