prusa3d/PrusaSlicer #3647: [PrusaSlicer 2.2.0-alpha4] - Does not start on Mac OS X 10.10.5 Yosemite

cdahinten, 2020-02-11:
### Version
[PrusaSlicer 2.2.0-alpha4] 

### Operating system type + version
Mac OS X 10.10.5 Yosemite

### Behavior
PrusaSlicers crasches on startup due to Invalid Code Sigature:

 Process:               PrusaSlicer [21679]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [21679]
User ID:               501

Date/Time:             2020-02-11 13:31:30.554 +0100
OS Version:            Mac OS X 10.10.5 (14F2511)
Report Version:        11

Crashed Thread:        0

Exception Type:        EXC_CRASH (Code Signature Invalid)
Exception Codes:       0x0000000000000000, 0x0000000000000000

kernel messages:

VM Regions Near 0x7fff736120a0 (cr2):
    __LINKEDIT             00007fff63983000-00007fff63997000 [   80K] r--/rwx SM=COW  /usr/lib/dyld
--> Submap                 00007fff70000000-00007fff80000000 [256.0M] r--/rwx SM=PRV  process-only VM submap
    unused shlib __DATA    00007fff721de000-00007fff747e2000 [ 38.0M] rw-/rw- SM=COW  system shared lib __DATA not used by this process

Thread 0 Crashed:
0   dyld                          	0x00007fff63914000 _dyld_start + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x000000000000000d  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x00007fff5b91eb68
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x00007fff63914000  rfl: 0x0000000000000201  cr2: 0x00007fff736120a0
  
Logical CPU:     0
Error Code:      0x020000f4
Trap Number:     133

Binary Images:
    0x7fff63913000 -     0x7fff6394992f  dyld (???) <3C068A0D-69D5-3AD6-86D9-6261D3B1564E> /usr/lib/dyld

kocikdav, 2020-02-11:
We are currently investigating this. Was 2.2.0-alpha2 starting normally for you?

cdahinten, 2020-02-11:
Hello kocikdav,

2.2.0 alpha2 is currently running fine.

Kind regards
CD

bubnikv, 2020-02-18:
This seems to be the same issue as https://github.com/prusa3d/PrusaSlicer/issues/3602

bubnikv, 2020-02-20:
Starting with 2.2.0-alpha1 we are using a new build server and new certificates. For some reason signing with these certificates seemed to work, but the signature was rejected by OSX 10.9.5, 10.10 and possibly by some other old OSX versions. We don't know what went wrong.

Anyway, we have a yet new build server freshly set up and quite powerful, and this time the certificates seem to be applied correctly, so they are accepted even by OSX 10.9.5.  I have updated the DMG archive at https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha4, please download https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.2.0-alpha4/PrusaSlicer-2.2.0-alpha4+-202002201411.dmg and test it.

Closing.

closed 2020-02-20 by bubnikv

cdahinten, 2020-02-20:
Hello Vojtěch,

tested: Works perfect now.

Thank you.
CD