prusa3d/PrusaSlicer #3648: Slicer missing large part of model

wpegden, 2020-02-11:
### Version
Tried with 2.2.0alpha2 and 2.2.0alpha4

### Operating system type + version
Ubuntu linux 18.04

### 3D printer brand / version + firmware version (if known)
MK3S, but I haven't printed this yet, as there is already a problem in the slicer.

### Behavior
A significant chunk of the model is missed by the slicer (the bottom section of the top-right quadrant of the heart.)  The model does not actually appear to be thin at this location.  This can be confirmed by separting the model into separate objects and examining that object separately.  After moving this part over and attempting to slice, Prusaslicer complains of no layers, which may or my not be related.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[upload.zip](https://github.com/prusa3d/PrusaSlicer/files/4186099/upload.zip)
[upload.zip](https://github.com/prusa3d/PrusaSlicer/files/4186100/upload.zip)
[image]
[image]
[image]
[image]

wpegden, 2020-02-11:
I have found another linux machine where the same appimage slices this file correctly.  However, both exhibit the seemingly related problem that an error message is thrown if attempting to slice after applying the "Split to objects" command.

wpegden, 2020-02-11:
One more comment: I have gotten the problem to happen on both machines now.  If someone else reports back on whether they see the same issue, I can work on helping with reproducibility.

I used the appimage of PrusaSlicer in both cases.

BeldrothTheGold, 2020-02-12:
PS complains this model has some issues when importing it into a Windows 10 build of PS. 
Regardless, it is slicing it properly. 

Also, when splitting the model into objects, several of the objects are empty. This causes the error your are seeing. If you delete the empty parts in the parts list, it will slice without complaint. 

[image]
[image]
[image]
[image]

BeldrothTheGold, 2020-02-12:
Not able to reproduce the slicing issue on my linux box either. Not with the latest AppImage nor a build from master.  Strange

[image]

BeldrothTheGold, 2020-02-12:
Oh, now that I think about it, maybe it's because the model is being fixed when it's saved as a 3mf. I don't believe it's possible to save a non-manifold model as a 3mf. 

If you open your attached 3mf are you seeing the issue?

wpegden, 2020-02-12:
That was a good thought!

If I open the 3mf file, I don't reproduce the problem.

But if I open the 3mf file, delete the object, and reload the heartbox.stl file (attached) and the slice, I get the problem again. 
[heartbox_stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4195600/heartbox_stl.zip)

BeldrothTheGold, 2020-02-13:
Ahh. Yes so the issue is caused by a non-manifold STL.  I know the PS team provided the repair feature in the file-menu to workaround slicing issues caused by non-manifold meshes. 

If you are OK with using those tools, would it be OK to close this issue?

wpegden, 2020-02-13:
Well to me it seems like there is a problem, even though there is a workaround.

When I add the heartbox.stl file, it shows up on the plate and under Info it says "Manifold: Yes".  It slices incorrectly missing a large part of the model.

If I save it as a 3mf file AND reload the file, it then says "Auto-repaired: 448 errors" under Info, and slices correctly.

Isn't it a bug that it is reporting "Manifold: yes"?

wpegden, 2020-02-13:
Just to add, you can see the "Mainfold: yes" line in the screenshots in my first post; at this point the slicer thinks STL file is fine, but as a result somehow has a slicing error.  As you see in the screenshots you posted, at that point the Slicer has realized the STL has a problem, and corrected 448 errors.  Shouldn't it realize from the beginning that the STL file is not a manifold?  Or at least not incorrectly report that it is?

BeldrothTheGold, 2020-02-13:
I suppose that is fair. The dev team will likely need the STL file to investigate further. 
If you are able would you please provide the zipped STL file?

wpegden, 2020-02-13:
I attached it to the comment above, which started "That was a good thought!"

wpegden, 2020-02-15:
Should I close this and open a new Bug report, under a simpler heading like: "Slicer thinks a model is a manifold when first loaded, but later finds and corrects 448 errors?"

closed 2026-09-29 by prusaslicer-bot[bot]