prusa3d/PrusaSlicer #3650: Command line SLA slicing

mmiscool, 2020-02-11:
### Version
I am using PrusaSlicer-2.1.1+linux64-201912101511 
have also tried in 
PrusaSlicer-2.2.0-alpha4+linux-x64-202002101524.

### Operating system type + version
Debian 9

### 3D printer brand / version + firmware version (if known)
Not applicable 

### Behavior
Running the following command results in a .gcode file. Renaming the .gcode file to .zip allow the archive to be opened and viewed. All png images are the same size and contain nothing but black. 

on 2.1.1 i use the following command line
`/prusa-slicer --export-sla --support-material --pad-brim-size 3 --layer-height 0.4  3DBenchy.stl `

on 2.2.0 i use the following command line. Hadded to add the --pad-brim-size 3 to make it work otherwise it complains. 
`./prusa-slicer --sla --support-material --pad-brim-size 3 --layer-height 0.4  3DBenchy.stl`

Results are as follows. 
[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/4188736/3DBenchy.zip)

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Original stl file 

[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/4188762/3DBenchy.zip)

BeldrothTheGold, 2020-02-12:
It would seem the command line isn't arranging the model before attempting to slice as manually setting a center works. 

`./prusa-slicer --center 50,30 --sla --pad-brim-size 3 --layer-height 0.05 ~/3dprinting/3DBenchy.obj -o ~/3dprinting/3DBenchy.zip`

generates the below

[image]

mmiscool, 2020-02-12:
Thanks @BeldrothTheGold 

Thjat seems to have solved my issue.

BeldrothTheGold, 2020-02-12:
@enricoturri1966 I don't see an -a or --arrange option in the command line help. 

Is the command line tool supposed to center the model inside of the print surface prior to beginning slicing? Much like the arrange tool inside of of the GUI.

enricoturri1966, 2020-02-13:
@BeldrothTheGold
There is no "arrange" option when running from command line.
Object positioning, from command line, for sla works the same as for fff.

tamasmeszaros, 2020-02-13:
As of now, PrusaSlicer's command line mode resembles very closely the original Slic3r. Your STLs should be auto-arranged, unless `--dont-arrange` is specified. The problem is with the settings. In command line mode, the installed printer profiles are not loaded automatically. Unspecified parameters are initialized with the hard-coded defaults, not with the Prusa SL1 defaults. So, even if you specify `--export_sla`, you don't have the proper bed geometry, output file format (extension) set. The tool has to know your printer in order to produce the right results. It is best to specify these settings as described in the manual https://manual.slic3r.org/advanced/command-line.

commit 726f645f (2020-02-18): Fix default hard coded parameters for SLA

fixes #3650
closed 2020-02-18 by tamasmeszaros