prusa3d/PrusaSlicer #3651: [Feature] Measurement tool / Ruler

ModischFabrications, 2020-02-12:
### Behavior
* _Describe the problem_

I need to validate a few sizes whenever I print a design made by somebody else, e.g. hole diameters.
Right now I need to start Meshmixer to make measurements, which makes iterations cumbersome, especially if I need to scale something (based on these measurements).

I feel like a simple ruler or measurements tool would allow users to switch less between programs. 
My most used measurement types in Meshmixer are:
1. (click) point normals in (measure wall thickness)
2. point normals out (measure diameters and fits)
3. generic point to point (fallback if 1 & 2 fail)

This could also lead to a "scale until dimension is X mm", but I don't want to get greedy, measurements are already a big step.

_Is this a new feature request?_ **Yes**

ModischFabrications, 2020-02-12:
Looks like someone has thought of a similar idea https://github.com/prusa3d/PrusaSlicer/issues/1655. I didn't find it due to the typo, but I feel like this one has a more extensive description.

danielkorczak, 2020-05-08:
I see both issues are closed and refer to each other. While I understand meshmixer is a good resource for this, my workflow would benefit from only needing one piece of software. Currently I only use meshmixer as a ruler 😁 

Can we continue the discussion on the merits of having a measurement tool?

pedjas, 2020-05-11:
I second this proposal. PrusaSlicer only shows total dimensions of an object. There is need to be able to measure specific part of the object to make size adjustments.

adam-wood, 2020-05-29:
Seconding this. Especially when scaling objects, it gets cumbersome having to measure elsewhere, then fiddle with scale calculation across two programs.

foreachthing, 2020-05-29:
@ModischFabrications : I got a question for you:
HOW do you measure a diameter that is actually no diameter? It's all triangles... It's not like a CAD where you can measure a diameter. So, how accurate do you expect your measurement to be?
Would it not suffice to display a scale on screen (like ANSYS does) to guesstimate the size of an object/feature?
See the scale (which changes according to camera-distance (zoom factor)) at the bottom of the model view.
[image]

And then again, how about you make your bed-shape svg like a millimetre-paper?

pedjas, 2020-05-30:
Almost each time I need to rescale object it was based on some feature on the object that has to match exact size, not dimensions on whole object.

foreachthing, 2020-05-30:
>.. I need to rescale object it was based on some feature on the object that has to match exact size, not dimensions on whole object.

For this, I use Blender with the 3D-Print addon.... Scale, click export, F5 in PS. Done.

pedjas, 2020-05-30:
I believe this is about PrusaSlicer, not Blender.

foreachthing, 2020-05-31:
I know. But why should someone reinvent something that already exists and works...? Next thing, someone wants to model directly in PS. This is just my opinion!! ;-)

ModischFabrications, 2020-06-01:
Sorry, late to the discussion, I was on vacation.

@foreachthing I would propose to measure diameters similar to the way meshmixer handles it with "Normal to face -> next intersection". I don't think that the precision "lost" due to the mesh is that important, everything is better than guessing by eye. 

I like your idea to show a millimetre-grid on the bed, but I think that is outside of the scope of this discussion. 

I agree that it's easy to overload a tool, I feel the same way about a lot of tools, but this case I would disagree with you. I feel like measurements are a core feature that is part of the workflow of many users. Sure, it's possible to do this blender or meshmixer, but something used that closely to the slicer feels fair to be included into it.

Let's not start a big discussion about the "need" for this feature until the development team has voiced their opinion, they are the ones that will decide if it's worth it or not.

pedjas, 2020-06-08:
I agree that application should not be overloaded with options but this is something what is needed. I believe this is necessary option in PrusaSlicer. 

Common work path is download STL from somewhere, load it in PrusaSlicer, slice and print. It is often needed to adjust print size and having just outer dimensions of the object is not enough.

It would be gret if user could set one or more measurement lines on object and that would update values as user changes object dimensions.

Ideally, user would be able to enter new value for dimension line and have whole object resized to match that.

peros550, 2020-06-13:
+1 this is extremely needed

git-jonathankinney, 2020-08-13:
I arrived here after searching for this feature, assuming I was missing something that already existed.  I would vote for this as well.  In memory objects have coordinates, that translate to dimensions in mm, that is how we know external object extent width, so it does not sound too far out of the scope of a slicer.  Its one step away from printing, where often it is not your model, and the description is lacking about inner diameter or some such dimension.  A measure tool seems like a reasonable request, not outside of scope for a slicer.  Like everything, this is a request for a feature, it requires someone to dedicate time and effort to make it happen, who knows what they are doing in this context, and currently that is not me.

Stroop14, 2020-08-18:
I agree that we need a quick basic measurement tool within prusaslicer to measure between two points for quicker resizing.

pedjas, 2020-09-05:
I just had situation where i needed measurement tool. I run slice on object with very small artifacts. So small that it could not match extrusion line width, so artifacts ended up smaller or larger than expected. 

If there is measurement tool in slicer I could check it before print.

lachyc, 2020-09-08:
It's cumbersome to have to open a model from Thingiverse in a modeller just to measure something rather than measuring and scaling in the slicer.

LongDono, 2020-09-08:
Came here via Google Search. It seems odd that you can adjust the scale of an item, with have the software know ing precisely the details of every part of the object, but you need to download another piece of software to measure between two points in order to get the scaling right. PrusaSlicer is by far a great tool that we know and love; this ability is just more of what you would expect from a Prusa product.

ssvenn, 2020-10-07:
I think one feature that would be extremely useful is the ability to measure the distance between two points, even if it's just on  a 2d plane, and then have the ability to automatically calculate a new scale-by percentage simply by adjusting the measured distance to your desired distance. It would be great for quickly tweaking an existing object to fit your custom need.

willbelr, 2020-11-07:
I use FreeCAD solely for this feature.. It would be so useful to do simple measurements directly in PrusaSlicer

smallouki, 2020-11-11:
Simple measurement between two points would be awesome indeed. Quick adjustments within STL are often required and it's most cumbersome to leave the slicer to do this.

jonnyhifi, 2020-11-14:
Developers time is limited. I would like a simple measurement tool that measures between the last two points clicked on, we as users can then use that data in many ways. (Ie not get bogged down with inferring wall thickness hole diameters etc).

The solidworks evaluate measure tool would be a good model - once measurement mode is selected (eg a button in the stack on the phd side of the screen "lay on base" "rotate object" etc), highlight the first point in the workspace highlighted with a mouse click, then it displays lines along x, y and z primary axis jointing the two points now highlighted, being measured on the model in the gui- with the distance for each primary axis dimension, and lastly a line for "direct distance" between the two point being labelled. Easy to implement as those different flavours of the same info can be so illuminating. (Sometimes one just wants to know "x width" as it were for example ... if one wants to check if holes or features are offset along x correctly for example) 

Solidworks measuring tool also decides how to handle what is clicked on from context - so such a measuring interface - easy to implement to start could be extended later if there was an inclination without rejigging  the initial interface (eg it thinking ... "oh" it's a circle I display diameter here ... "oh two planes ... I display angle between them" etc but I stress These complications I think would be lovely but very low priority as clearly non trivial with stl (rather than parametrised solid model) files to implement) so
Please don't discount this idea on the basis of the benefit / time taken for the advanced version isn't worth it but do please give us the basic version, with x,y,z and "as the crow flies" distance between two points.

QUAKEULUS, 2020-11-26:
How about a simple texture for the 'build plate' texture? instead of unmarked black lines on grey, have a ruler from the origin/center? as we zoom in numbers grow in precision on the build plate? This won't require any meshing

Zewwy, 2020-11-29:
> How about a simple texture for the 'build plate' texture? instead of unmarked black lines on grey, have a ruler from the origin/center? as we zoom in numbers grow in precision on the build plate? This won't require any meshing

What a great n simple idea for now, def beats adding complexity.

QUAKEULUS, 2020-12-06:
Where is the "bed" texture file in the directory?

foreachthing, 2020-12-07:
> Where is the "bed" texture file in the directory?

Check your "printer".ini file in your [profile folder](https://help.prusa3d.com/en/article/faq-prusaslicer_1789). There's a var called "bed_custom_texture".  Or, [create your own texture](https://help.prusa3d.com/en/article/custom-svg-and-png-bed-textures_124612) and place it in your profile folder, then access it from the printer-settings page.

QUAKEULUS, 2020-12-07:
> Check your "printer".ini file in your [profile folder](https://help.prusa3d.com/en/article/faq-prusaslicer_1789). There's a var called "bed_custom_texture". Or, [create your own [...]

Thanks! There's still a problem:

"If the texture size doesn't match the bed size, the texture will be stretched to perfectly fit the bed. It is not currently possible for the texture to have elements outside of the print area."

This means each bed size must have a custom texture.
An easy solution would be a checkbox to "center texture at origin"
and then have  a larger square texture that would be centered on any print area, and just cropped by any custom print bed size. 
Maybe I should make a separate thread for that as a feature request?

foreachthing, 2020-12-07:
Off topic:

> This means each bed size must have a custom texture.

Yes. But that is not a problem.
Here you can find graph paper in svg format: https://freesvg.org/graph-paper-sheet

> An easy solution would be a checkbox to "center texture at origin"
> and then have a larger square texture that would be centered on any print area, and just cropped by any custom print bed size.

I would not want that. Since my printer's zero is in the bottom left corner (my svg would need to be 4 times as big as the bed and would waste memory and resources). You'd need an offset (x, y) parameter to move the texture to the correct location and/or an option not to scale the texture at all.

> Maybe I should make a separate thread for that as a feature request?

Yes, that would be best.

andsk8, 2020-12-20:
Yo, this measurement tool is extremely needed. For example, to check if a hole is big enough for the purpose. Don't need to be fancy, just distance between 2 points. If easier, measurement could only be taken if view is fixed in an axis, so you don't need to compensate perspective. Thanks!

Ali229, 2021-01-29:
Extremely important feature for functional prints

poxlox, 2021-04-09:
This is definitely needed and I will explain why with the exact reason why I even found this thread.
I am trying to measure the internal points of a re-scaled object that I am merging with another object in order to create one complete object. In this case I am trying to measure a distance on one location on the model, where they aren't even the same .stl.

rananna, 2021-04-22:
I also would like to see this feature

cryoglobe, 2021-07-23:
Hello ...
I just wanted to push this "small" request  alive ..;  all the good reasons have already been given by the other posts/comments

Please , help us .. PrusaSlicer is great  ...  add this function and i will really save us lots of time/checks!!

poxlox, 2021-07-26:
I am still hoping to see this feature.

Hologos, 2021-08-16:
How come this wonderful piece of slicer doesn't have this essential tool? :o

tapaul50, 2021-09-23:
+1 - I would use this tool daily.  Thank you

jakubmirejovsky, 2021-09-25:
I just googled "PrusaSlicer measure" and hoped this was a core part of the tool. Well, some day maybe? :)

Akaj-lab, 2021-11-08:
After so long i can't believe this feature still doesn't exist
How???

Please, please, please

Sqeaky, 2021-11-12:
This is an obvious feature to have. I have a part and I need to know measurements on it. This software can change sizes of part, but not check results? Seems like an obvious oversight. 

 and I am unfamiliar with the other software in this thread, getting another piece of software

If I write this feature is the patch likely to be accepted?

I am imaging a button that you click in one of the toolbars then you click in 2 places on the part and produces a measurement and puts it in the status area. Is that a reasonable approach if I do decide to code this up?

bubnikv, 2021-11-12:
We plan to implement some measurement tool for 2.5.

> If I write this feature is the patch likely to be accepted?

I don't know. Depends on the code quality. Also because we have it on our list, it may be that you will develop the feature in parallel with us.

Sqeaky, 2021-11-12:
Then I will skip it for now.

I am familiar with C++17 but not WxWidgets or this codebase. If you didn't have plans I would do it, and if those plans fall through then me muddling through it is better than not having the feature. So let me know if that would be useful.

I would make sure the code is up to your standards and has tests. I downloaded the repo and can build it without error, Sorry If I sent in a bunch of hardware info accidentally. I was able to run some tests and they all passed. I only explored the code for an hour or two, I can make sense of it, but would need time to figure out a good way to do it.

jordylane, 2021-12-17:
What they said. ^^^

yanglu2021, 2022-01-12:
I just bought a Prusa printer. The first print was the whistle came from the SD card. Now, I am printing the second one I downloaded online and started to need a ruler tool to check hole distance, in which case google led me here. As everyone said, this tool is the top most used tool and I cannot believe it has not been there.

poxlox, 2022-01-12:
"need a ruler tool to check hole distance"
Most obvious use case. Coming up on a year since I first commented, and I still want to see this.
I'll tell you why it has to be in-app. SCALING.
Everybody scales their things, and it's much easier to understand how the scaling has affected your model by a ruler. In my case, I often combine 2 or more models, which can be a real nightmare to calculate minute features about it after I've settled on the various scales and layout. It would take MUCH longer to go back and calculate based off the original model and its scale, and even harder when I'm looking at various parts of the model rather than its entire dimensions.
Just, please, I want the tool.

ProjectCleverWeb, 2022-01-13:
This is a must-have feature!

NathanielArmer, 2022-01-17:
Found this because I desire this feature, downloaded a STL for holder that some claim might not fit the desired object. A ruler would allow me to quickly double-check the design.

russiancosmo, 2022-01-26:
This is a must-have feature! Cura offers a plugin-measuring tool so I think it is doable for Prusa as well...

BlackBird11, 2022-01-27:
would also love this feature! please!

fogarasia, 2022-02-02:
Waiting for this great feature!

ozgurgesli, 2022-02-05:
I needed this feature (any point to point measurement is fine) too.

ModischFabrications, 2022-02-05:
OP here: I'm happy to see everyone else sharing my opinion, but I think that more "I want it too" comments won't help. Feel free to "+1" the original comment instead to reduce notification spam for subscribers :)

islander2013, 2022-02-25:
lets add this useful tool to a great piece of software!

rkarlsba, 2022-04-08:
I'd love to see this feature. One thing is to use other software like meshlab to measure the STL, but if this was integrated in PS, it'd be a lot easier to resize parts etc.

sergioprusky, 2022-04-26:
A simple measuring tools would be useful. Printing an adapter for 3/4" PVC, want to make sure that it is the correct size before I print it. Millimeters is fine, converting to/from inches is simple enough.

dwongdev, 2022-05-03:
Would love to see this feature added.

gabriele-cannizzaro, 2022-05-28:
Big thumbs up for this feature!

dohlin, 2022-07-10:
Would really like to see this implemented...it's sorely missed coming from Cura.,

Dummy1234567891, 2022-07-20:
There must be a reason they haven't added it yet..maybe because it won't increase sales so not worth doing or something...

stepikovo, 2022-07-20:
@Dummy1234567891 PrusaSlicer is for free :) 

As already mentioned, we are working on that tool, might be in 2.6.0, no need to "vote" for it anymore

ygoe, 2022-07-24:
Any workarounds or other tools for this? I downloaded a model and need to measure its diameter to verify if it's of any use for me. The publisher didn't care to describe the technical details of the model.

haarp, 2022-07-24:
> Any workarounds or other tools for this? I downloaded a model and need to measure its diameter to verify if it's of any use for me. The publisher didn't care to describe the technical details of the model.

You can create a modifier which you then scale/move until it fits precisely into the feature you want to measure, then read its dimensions from the sidebar. It's tiresome, but it does work.

[21 more replies]

closed 2023-02-01 by lukasmatena