prusa3d/PrusaSlicer #3652: Stray Prime Tower Extrusions

Lukrative525, 2020-02-13:
### Version
2.1.1

### Operating system type + version
Windows 7 Professional

### 3D printer brand / version + firmware version (if known)
Custom i3 running RepRapFirmware. I haven't gotten past the slicer with this one, so this should be irrelevant.

### Behavior
By the title you might think this is an uninformed repeat of issue #1070, but the behavior is substantially different.

I load a model, enable soluble support and the prime tower, and position the prime tower. The render shows intermittent extrusion lines coming off the prime tower going in the negative x-direction, generated seemingly at random heights.

* _Expected Results_
The slicer should not generate any extrusion lines other than those found in the prime tower, support, model, skirt, etc.

* _Actual Results_
The slicer generates extra extrusion lines which serve no purpose and could potentially pass over the model.

[image]

#### Project File (.3MF) where problem occurs
[3DBenchy dual extrusion test.zip](https://github.com/prusa3d/PrusaSlicer/files/4197001/3DBenchy.dual.extrusion.test.zip)

Thanks y'all!

lukasmatena, 2020-02-13:
This should be already fixed in 2.2.0-alpha4 (#3473, #3234). Can you retest?

Lukrative525, 2020-02-14:
The problem seems to be fixed. Thanks!

closed 2020-02-14 by lukasmatena