prusa3d/PrusaSlicer #3653: [PrusaSlicer 2.2.0-alpha4]

tilllt, 2020-02-13:
### Version
PrusaSlicer 2.2.0-alpha4

BTW, the Copy to Clipboard Option in the About Panel doesnt work

### Operating system type + version
OSX 10.12.06

### 3D printer brand / version + firmware version (if known)
Creality Ender3

### Behavior
The Alpha 4 doesnt save user settings on my system. Even when starting with the Wizard from Scratch, deleting old profiles, after closing the Program all custom settings are gone.

FidelCapo, 2020-02-14:
Thank you for report. You are right. Copy to clipboard does not work on mac. And filaments cannot be selected when custom printer is selected in Configuration wizard. We will fix it.

Could you please describe "The Alpha 4 doesnt save user settings on my system" more? Best to write all the steps to reproduce issue. Thank you.

bubnikv, 2020-02-18:
> Copy to clipboard does not work on mac. 
At which dialog? The "About PrusaSlicer" or "System Info"?

We would  be thankful for a detail description as of what steps did you perform and what went wrong.

tilllt, 2020-02-19:
Sorry for the slow reply. I am not in my studio at my prusaslicer & 3d printer so i will try my best recollect what happens. Basically the Alpha4 didnt save any ANY settings, even when explicitly saving the changes. On Restart, usually the entire preset for the Ender3 and all Filaments were gone, even the pre-installed. I found a workaround by constantly doing config snapshots, which in fact are saved correctly and can be loaded. When i am back at my Mac & Printer, i will try to write a more detailed report.

As an amateur guess, to me it looked like a permission error, where prusaslicer couldnt save anything but also didnt complain about it.

tilllt, 2020-03-04:
As far as I can see, 2.2.0-rc does not show this behaviour anymore, seems to work now.

closed 2020-03-04 by tilllt