prusa3d/PrusaSlicer #3654: Linux Mint 19 unable to launch slic3r-prusa

eiger3970, 2020-02-13:
### Version
1.39.1+dfsg-3

Version was in the Software Manager Launcher.

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Linux Mint 19 Tara 64-bit
Unable to Launch.

### 3D printer brand / version + firmware version (if known)
Anycubic Photon

### Behavior
Linux Mint > Panel > Menu > System > Software Manager > Search: prusa > Slic3r-prusa > Install > Launch > Panel > Menu > Applications > Programming > Slic3r Prusa Edition > nothing opens.
Uninstalled and reinstalled and same issue.

_Is this a new feature request?_
No.

I ran from Terminal and the error output is:
`$ slic3r-prusa3d --gui
Can't locate Wx.pm in @INC (you may need to install the Wx module) (@INC contains: /usr/bin/lib /usr/lib/slic3r-prusa3d/ /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.26.1 /usr/local/share/perl/5.26.1 /usr/lib/x86_64-linux-gnu/perl5/5.26 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.26 /usr/share/perl/5.26 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at /usr/lib/slic3r-prusa3d/Slic3r/GUI/2DBed.pm line 10.
BEGIN failed--compilation aborted at /usr/lib/slic3r-prusa3d/Slic3r/GUI/2DBed.pm line 10.
Compilation failed in require at /usr/lib/slic3r-prusa3d/Slic3r/GUI.pm line 9.
BEGIN failed--compilation aborted at /usr/lib/slic3r-prusa3d/Slic3r/GUI.pm line 9.
Compilation failed in require at (eval 91) line 1.`

lukasmatena, 2020-02-13:
The problem indicates there are some Perl modules missing. Although I am not a Linux expert, I think this should have been taken care of by the packaging system. However, I don't think it is worth investigating - version 1.39 is ancient anyway.

Unless you have a serious reason not to, I suggest using current PrusaSlicer 2.1.1 that you can download as an AppImage here: https://github.com/prusa3d/PrusaSlicer/releases. You should face no Perl-related problems there and get an application that is much much more evolved.

eiger3970, 2020-02-13:
Thanks.  Just tried AppImage and posted this issue: https://github.com/prusa3d/PrusaSlicer/issues/3659

bubnikv, 2020-03-15:
1.39.1 is history.
If it is a distro package, you should consult with your distro package manager.
Closing.

closed 2020-03-15 by bubnikv