prusa3d/PrusaSlicer #3655: Opening saved project removes Print Host settings

brettvitaz, 2020-02-13:
### Version
2.2.0-alpha4

### Operating system type + version
macOS Catalina 10.15.3

### 3D printer brand / version + firmware version (if known)
Creality Ender 3

### Behavior
Saving a project and opening same project causes "Print Host upload" settings to get removed. 

Steps to reproduce:
* Have printer profile with OctoPrint configured
* Add model to build plate
* Save project
* Open saved project
* Navigate to Printer Settings / General

Expected:
* Hostname, API Key are filled
* Slicing gives option to send to print host
[image]

Actual: 
* Hostname, API Key are blank
* Slicing does not give option to send to print host
[image]

#### Project File (.3MF) where problem occurs
[Stamp Assembly.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4199381/Stamp.Assembly.3mf.zip)

Jebtrix, 2020-02-13:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3244
See https://github.com/prusa3d/PrusaSlicer/issues/3050 for workaround
See https://github.com/prusa3d/PrusaSlicer/issues/3642 for recent (minor) UI side effect

brettvitaz, 2020-02-13:
Thank you, the workaround is working.

closed 2020-02-13 by brettvitaz

bubnikv, 2020-02-18:
patched with 25d6818fc7351ff008674adb8ff15e9522246764

When loading a configuration file (from AMF, 3MF, .ini or .gcode), 
and if the host settings ("print_host", "printhost_apikey", "printhost_cafile")
in the configuration file is empty, then the config is considered to be anonymized,
and these anonymized keys are loaded from the referenced "derived from" profile,
if it exists in user's PrusaSlicer configuration directory.

We are aware that this is a patch work and a better long term solution
is to separate physical printer settings from logical printer settings,
but this is a good enough solution for PrusaSlicer 2.2.0 release.

The patch will be part of PrusaSlicer 2.2.0-beta. Closing.