prusa3d/PrusaSlicer #3656: incorrect visual appearense & or incorrect sclicing in all

weekendkoder, 2020-02-13:
### Version
PrusaSlicer vers. v2.10 stable and v2.20  alpha

### Operating system type + version
Windows 8 and Windows 10

### 3D printer brand / version + firmware version (if known)
JGaurora A5 , firmware: 1.18c Marlin
Tronxy X5SA , firmware: chituWare

### Behavior
When you Slice, you can see the progress of the slice. First all is visually displayed in yellow. In the final stage it has colors, depending on the "legend" feature type. And this is what happens, when i slice my object : 
 
1. first stage (yellow) looks correct, everything is fine
2. second stage (colored) looks incorrect, because lines are missing. Only there, where the diameter is about 0,4 - 1,6mm. (in my case it is a support structure) 

check the picture, which is included in this post. You will see the missing areas. The strange thing here is : 
1. first stage is not affected by this error, only the second final stage
2. it depends on how big the STL is, you load in. I tried also a cutted section of the object in which the error appears. If its smaller, the error dont appear. And also if its a totally different object (which does not contain a tree-support).

But about all, it has its trouble with very small diameters, like 1,2mm or 2mm. Which you maybe usually use by a tree-support. And only than, if its a "heavy geometry" you need to slice. Small objects seems not to have this problem.

* _Steps needed to reproduce the problem_
Just load some big heavy geometry into prusaSlicer. Before that make some tree-Supports (maybe inside MeshMixer) with about 0,6-1,6mm. And it should be reproduceable.
 
#### Project File (.3MF) where problem occurs
Can´t provide a upload. Because it is confidential material. I hope you got it through my describing and the picture.

Jebtrix, 2020-02-13:
Have you tried using all default comparable prusa printer profiles? I would try to narrow it down to your custom profile settings first.

enricoturri1966, 2020-02-14:
@weekendkoder
Is this issue happening with PrusaSlicer 2.2.0.alpha4 ?
There was a commit (https://github.com/prusa3d/PrusaSlicer/commit/63f8fda61d7222ca2e93aad8e47d32025e701ac5) that may have already fixed it

weekendkoder, 2020-02-14:
#defaul printer profiles
Ehm , i guess i tried it. But with no success

#Prusa 2.2.0 alpha4
Yes. I tried also this version. No success

I was able to modify my object. So i can upload it here. You will have the *.3mf file and the *.stl file.
It would be great if someone of you can investigate into this problem and maybe confirm the error.
Thank you.

Link to the files:
https://github.com/weekendkoder/PrusaSlicer-Test-Files

Jebtrix, 2020-02-14:
Print Settings > Custom G-Code > Before Layer Change G-Code

Slicing works fine if you add this (from default Prusa printer profile)

```
;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]
```

weekendkoder, 2020-02-15:
#jebtrix
Thats it ! This solve my issue with the print. Thank you very much for this important hint and your rapid answer !

closed 2020-02-20 by lukasmatena