prusa3d/PrusaSlicer #3659: AppImage won't launch

eiger3970, 2020-02-13:
### Version
PrusaSlicer-2.2.0-alpha4+linux-x64-202002101524.AppImage

### Operating system type + version
Linux Mint 19 Tara 64-bit
Kernel Linux 4.15.0-76-generic x86_64
MATE 1.20.1

### 3D printer brand / version + firmware version (if known)
ANYCUBIC PHOTON.

### Behavior
* AppImage won't launch.  Error: Could not display "/home/linuxmint/Downloads/Pr...nux-x64-202002101524.AppImage".  There is no application installed for AppImage application bundle files.  Do you want to search for an application to open this file?
* Releases · prusa3d/PrusaSlicer > Assets (down to middlish of page) > select PrusaSlicer-x.x.x-alpha4_linux-x64-YYYYMMDDmmhh.AppImage > Downloads directory > right click PrusaSlicer-x.x.x-alpha4_linux-x64-YYYYMMDDmmhh.AppImage > Properties > Permissions > tick Allow executing file as program > Close > double click AppImage.

Strangely, the AppImage open the first time without issue.
I walked through the welcome wizard, however the welcome wizard was irrelevant to my SLA 3D printer.  I had set FDM extrusion to None, however the wizard then asked for extrusion widths.  I entered 0 mm and finished the wizard.
The AppImage could no longer open, as it said something like, extruder widths were too small.
I deleted the AppImage, redownloaded and tried to open and the posted error occurred.

Area5142, 2020-02-14:
To completely reset and restart with welcome wizard the directory _~\.PrusaSlicer-alpha_ had to be deleted. Alpha version stores configuration here.

I agree a extrusion width of 0 mm should not prevent application startup.

eiger3970, 2020-02-15:
Wow, PrusaSlicer is brutal.  In Configuration Wizard, I tried to setup an SLA printer.
I selected the SLA printer, but nothing for the ANYCUBIC PHOTON.
In the FFF printers, I selected None.
I then Finished the wizard and it shutdown with a message something like: are you extruder sizes too small?
I then tried to reopen the appImag and error:
Internet error: Flow::spacing() produced negative spacing. Did you set some extrusion width too small?

bubnikv, 2020-02-18:
> I then tried to reopen the appImag and error:
> Internet error: Flow::spacing() produced negative spacing. Did you set some extrusion width too small?

This is a similar issue to https://github.com/prusa3d/PrusaSlicer/issues/3482, which I have fixed today with 43f2171446a3de8b88b8ad20c1df71a96b7c4341.

We should also look into the wizard, but I believe this fix is sufficient to at least keep the application alive.

FidelCapo, 2020-02-21:
This issue should be fixed in next release.
Closing.

closed 2020-02-21 by FidelCapo