prusa3d/PrusaSlicer #3660: Slicing Error modal window and Context Menu blocks user input totally.

imonsei, 2020-02-14:
### Version
2.1.1

### Operating system type + version
windows 10

### Behavior
* _Describe the problem_
If you have "background processing" enabled, and bring up a context menu (ie, right click a model), while it is slicing in the background, and the slicer encounters a problem, which brings up the error modal window, but you cannot click on anything in the program.

* _Steps needed to reproduce the problem_
1. enable "background processing" in preferences
2. import a stl file with errors in it, for instance a model with details so tiny they can't be printed.
3. click "ok" when the error message modal window pops up.
4. move the model (so it reslices) and while it is reslicing, right click your model to bring up the model's context menu.
5. context menu is now visible, and error modal window pops up, but is not clickable.

* _Expected Results_
Clicking "ok" should close the modal window.

* _Actual Results_
Not being able to click on anything in the prusaslicer program, not even the close button.
You are not able to alt-tab with the keyboard.

### Workaround
If you click any other window, or windows interface object (like the taskbar), removes focus from the prusaslicer program, and the context menu disappears, making you able to click the ok button on the modal window.

JohnnyDeer, 2020-03-12:
Hi, thank you for reporting. Unfortunately i'm not able to reproduce it. Can you please provide stl wich makes this error and ideally screenshot of that problem situation?

bubnikv, 2020-03-12:
Do you get it always, or just with some models?

imonsei, 2020-03-12:
Hi guys. 
Attached is a screenshot of the bug happening.
For testing I made 2 cubes in blender3d, moved one above the other and made sure they weren't touching eachother, so we have a nice gap in between. Then I joined them with ctrl-j and exported the model as stl.
Load the model in prusaslicer. click ok when the error window pops up. move the model with the mouse and immediately right click it, to bring up the context window, and wait for the background slicing process to finish, which will pop up the error modal window.
[image]

I understand 2.2.0 is right around the corner, so I'm gonna hold off further testing until it's out. If it's not present in 2.2.0 I'll close this bug report.

JohnnyDeer, 2020-03-12:
Thank you for describing, now we are able to reproduce it.

JohnnyDeer, 2020-03-12:
I provide STL to reproduce error...

[#3660 Slicing Error modal window and Context Menu blocks user input totally.zip](https://github.com/prusa3d/PrusaSlicer/files/4324518/3660.Slicing.Error.modal.window.and.Context.Menu.blocks.user.input.totally.zip)

bubnikv, 2020-03-12:
Fixed with 1d5249f by postponing the message box after the pop-up menu closes.
The fix will be part of PrusaSlicer 2.2.0-final.
Closing.

closed 2020-03-12 by bubnikv