prusa3d/PrusaSlicer #3661: issue with printer pause

tr3topflyer, 2020-02-14:
### Version
alpha 2.2.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Prusa MK3 3.8.1 2869
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
When inserting a "pause" from peslicr, the printer does pause and the print head moves out of the way.  On the printer display it shows that the printer is paused but there is no way to make it resume.  Pressing the control knob does nothing.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2020-02-14:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3557

lukasmatena, 2020-02-18:
As mentioned by @Jebtrix, this is a duplicate of #3557, the problem was in FW and it should already be fixed. Closing.

closed 2020-02-18 by lukasmatena

stedoc, 2020-03-08:
The problem was solved with update fw, but the text appears only for a few seconds

bubnikv, 2020-03-09:
the firmware team is aware of the text issue.