prusa3d/PrusaSlicer #3662: {Feature MMU2: Ramming behaviour in single mode

electrickery, 2020-02-14:
### Version
2.1.1+linux64

### Operating system type + version
Xubuntu 18.04 LTS

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S 3.8.1-2869 + MMU2S 1.0.6-372

### Behavior
Ramming behaviour is only enabled for more than one extruder, which also introduces a wipe tower. currentlt I mainly try to use the MMU2S as a easy way to use different filaments for different prints, so just as a fast filament change device. But because no ramming is employed in this mode filament loading is not very reliable.

_Is this a new feature request?_ Yes

#### Project File (.3MF) where problem occurs
The problems arise during filament loading to the extruder, so the actual model is not very relevant, other than it being single filament.

wpegden, 2020-02-14:
This is also an issue for me.

Most of my failed loads are first or last loads, which are not as annoying as other loads, but should be preventable.

bubnikv, 2020-02-18:
What is not reliable? Filament loading or unloading? How does the problem
manifest? What do you recommend?

electrickery, 2020-02-18:
Thanks for your enquiry.

The problem arises during the load to extruder. In MMU2 single mode, filament is not treated with the ramming procedure while unloading, so the end is quite stringy (working mainly with PLA here). When the same filament is loaded again to the extruder, it is very likely to fail.

My recommendation would be to make the ramming procedure an option in single mode, making the usage of filament change here as reliable as in full multi-material mode. 

The overall usage mode is having the most used four or five filaments loaded to MMU2 permanently and select one per print.

bubnikv, 2020-02-18:
I see. I believe there is some ramming going on, which is either scripted
in the custom G-code of the printer profile, or in the printer firmware,
but it may not be optimal.

gag, 2022-11-19:
I do have the same issue. Would love to have this implemented!

dukes4, 2023-01-15:
would be great to get this solved - please

closed 2026-09-29 by prusaslicer-bot[bot]