prusa3d/PrusaSlicer #3663: User settings (Print, Filament, Printer) not showing, working in 2.2.0a2

skohls, 2020-02-15:
### Version
2.2.0a4

### Operating system type + version
OS X 10.15.3

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S; Prusa MK3S

### Behavior
The user profiles for Print, Filament and Printer don't show up on 2.2.0a4. 
I've removed any dependencies I could find and think of in the .ini files already.
In 2.2.0a2 it works and in 2.1.0 too

### Feature request
Also, with all the crazy possibilities regarding dependencies - perhaps a simple tool to just manually create the relations of ones .ini files would be in order. Or at least a setting to ignore any restrictions and give the user a chance to get the stuff together. =^..^=

I've attached the current config bundle I use in 2.2.0a.
[PrusaSlicer_config_bundle.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/4208381/PrusaSlicer_config_bundle.ini.zip)

FidelCapo, 2020-02-17:
Unfortunately, I am not able to reproduce this issue. Can you please describe all steps to reproduce the problem? Thank you.

skohls, 2020-02-17:
Well, when I first installed 2.2.0a2 it created a new folder for its settings. I copied my custom .ini files from 2.1.0 over to this -alpha folder and it all worked well, I could use all my custom print/filament/printers settings I was used to.
When I start 2.2.0a4 it uses the same -alpha folder but will only show a very small set of combinations from my custom settings. Mostly the filaments are not showing up.
I'll try to do some more screenshots when I'm home later today.

neophyl, 2020-02-17:
Looks like a duplicate of #3658 and also #3347 which was closed 25 days ago as being fixed in a comit.

skohls, 2020-02-17:
OK, I've checked the linked issues and was able to get my profiles to like my filaments again by making sure they had "inherits = 0.20mm QUALITY @MK3"  in them
Did this to all my print profiles and 2.2.0a4 now shows all the filaments again.

Thanks for a hint in the right direction!

closed 2020-02-17 by skohls

bubnikv, 2020-02-18:
> Also, with all the crazy possibilities regarding dependencies - perhaps a simple tool to just manually create the relations of ones .ini files would be in order. Or at least a setting to ignore any restrictions and give the user a chance to get the stuff together. =^..^=

By the way, there is a "Show incompatible print and filament presets" checkbox at the Perferences dialog. If you check that checkbox, then there will be a red / green flag shown next to the combo box with the profile selection. 

[image]

Click on the flag to turn it red and then all the profiles are shown, even the incompatible ones. This is considered a debugging tool for the designer of profiles, not a production tool.

skohls, 2020-02-18:
Ah, thanks for that heads up - I missed that one completely!

neophyl, 2020-02-18:
Except as I mentioned in issue #3347 before it was closed the Preferences option to show incompatible profiles does NOT work with the new system.  Previous printer profiles created with older versions of Prusa Slicer and imported into alpha 4 (even when based on an existing MK3 profile) when selected are missing the default filament and print profiles unless edited with the work around.