prusa3d/PrusaSlicer #3665: "Support material extruder" setting affects preview but not gcode

wpegden, 2020-02-15:
### Version
2.2.0 alpha4
### Operating system type + version
Ubuntu

### 3D printer brand / version + firmware version (if known)
MK3S+MMU2S

### Behavior
Under print settings->multiple extruders, we find the settings (among others)
"Support material/raft/skirt extruder"
"Support material/raft/interface extruder"

In the SOLUBLE INTERFACE profile, these are set to 0 and 5, respectively, while in the SOLUBLE FULL profile, these are set to 5 and 5, respective.

As expected, the SOLUBLE FULL profile uses only extruder 5 (which should be the soluble filament) for supports.

The behavior I expected from the SOLUBLE INTERFACE profile is that it would only use soluble material at the interface, not for the supports.  This is not what happened.  Instead, the supports were printed with a mix of both soluble and regular materials (whichever was currently loaded).  Given that "Support material/raft/skirt extruder" was set to 0, this is perhaps not surprising.

HOWEVER, if I CHANGE this extruder setting to 4 (my PLA extruder for this print), then the preview changes to show supports printed only with PLA, with PVA interfaces, as you would expect, but the GCODE output still switches between PLA and PVA for the supports, as if 0 was still selected.  (To confirm this you can use a gcode viewer like the one bulit into octoprint which can show different extruders, or try printing the gcode file from the sd card.  For the purpose of testing this bug, it is not necessary to use soluble material in extruder 5, just a different color.)

I'm attaching two screenshots.  The first is what happens in the preview window when you slice with "SOLUBLE INTERFACE" profile without changing anything.  It shows it will use both materials for the support (which seems already like the wrong behavior given the point of this profile!).
[image]

The second shows the preview window after changing "Support material/raft/skirt extruder" to extruder 4.  However, as can be seen by examining the gcode file attached in the zip, it still mixes both filaments in the support, as in the first screenshot.

[image]

[test.zip](https://github.com/prusa3d/PrusaSlicer/files/4208986/test.zip)

lukasmatena, 2020-02-17:
Thanks, you're right. This was apparently broken between alpha2 and alpha4. We will fix that.

wpegden, 2020-02-17:
Terrific!

Is the intended behavior that it will mix soluble and nonsoluble material in the supports in the Interface profile unless this "support material extruder" is manually changed?  

I.e., is the bug just with the gcode not matching the preview after changing this setting, or also with the behavior when the preview and gcode disagree, before changing this setting?

wpegden, 2020-02-18:
To answer my own follow up question, in alpha2, the behavior seems to be to not use the soluble extruder (5) in the noninterface supports, even when it is set to 0.  (To me this seems like the right behavior!)  Glad to hear the problem in alpha4 will be fixed.

lukasmatena, 2020-02-18:
It should be fixed by b1145df for 2.2.0-beta.
Thanks a lot for reporting this.
Closing.

closed 2020-02-18 by lukasmatena