prusa3d/PrusaSlicer #3666: Error parsing Prusaslicer config file

Booner99, 2020-02-15:
Alpha 4

I have a error when trying to start Prusaslicer.

The error message is error parsing Prusaslicer config file

kocikdav, 2020-02-17:
Hello. Would you provide little more information please? What system are you working at? I assume this happened at the first run of 2.2.0-alpha4? What alpha version were you using before? Note that alphas are using different folder for config files than regular PrusaSlicer. Also was the error assuming any action to recover from the error? If so, did it help?

lukasmatena, 2020-02-17:
Apart from all the information requested by @kocikdav, if you're still in that state, it would be very helpful if you could zip your config directory and send it to us, so we can reproduce the problem and see what exactly went wrong.

The config directory should be in C:\Users/username/AppData/Roaming/PrusaSlicer-alpha (if you're on Windows).

bubnikv, 2020-03-15:
no response, cosing.

closed 2020-03-15 by bubnikv