prusa3d/PrusaSlicer #3667: super small "Send GCODE"? Missing shortcut. Still Loosing IP/API every loading 3mf..

3Dietrich, 2020-02-15:
### Version
2.2.0-alpha2+

### Operating system type + version
MacOS Sierra 10.12.6
### 3D printer brand / version + firmware version (if known)
i3Mk3
### Behavior
The IP and API Key in Printer Settings are lost every time. I know that this was reported, I just want to enforce that problem and combine it with other connected problems.. 
You seem not to work via network (e.g. octoprint), else you would already fixed the missing ip/API I'm sure. Also, the Button to "Send" is now hardly visible anymore..? And there no Shortcut for Send. That together feels like you're working on 'SD card only' Version?

#### _Steps needed to reproduce the problem_
"Send button"
- search the "Send GCODE" button
- and try to find a shortcut for it

#### _Expected Results_
- IP/API should recall as I entered it every frqing time,
- "Send GCODE" should be bigger
and/or..
- create a Shortcut for "Send GCODE"
#### _Actual Results_
- it offers _blank space_ in IP and API Key
- "Send.." is a little unnamed Button right behind a long [_____________Export G-Code_____________] [ > ]
- no way to avoid mouse handling to the little right Play button because there is NO Shortcut

bubnikv, 2020-02-19:
> IP/API
> set IP adn API
> save the .3mf, load it again
> --> IP and API lost

This is a duplicate of:
    Opening saved project removes Print Host settings #3655
    Octoprint Settings when Opening a .3MF file #3244
and it has been fixed with 25d6818fc7351ff008674adb8ff15e9522246764:

    When loading a configuration file (from AMF, 3MF, .ini or .gcode),
    and if the host settings ("print_host", "printhost_apikey", "printhost_cafile")
    in the configuration file is empty, then the config is considered to be anonymized,
    and these anonymized keys are loaded from the referenced "derived from" profile,
    if it exists in user's PrusaSlicer configuration directory.
    
    We are aware that this is a patch work and a better long term solution
    is to separate physical printer settings from logical printer settings,
    but this is a good enough solution for PrusaSlicer 2.2.0 release.

bubnikv, 2020-02-19:
Shortcut for Send G-code is available for couple of versions.
[image]

bubnikv, 2020-02-19:
We will add the shortcut description 
into the tooltip of the "Send to printer" button.

YuSanka, 2020-02-19:
Thanks for your report. This bug is fixed and will be a part of upcoming release.

closed 2020-02-19 by YuSanka

3Dietrich, 2020-02-19:
> Shortcut for Send G-code is available for couple of versions.
> [image]

ah! damn, I looked in the **Keyboard Shortcuts** via ' ? ', it's missing there. Maybe you add this shortcuts to the list of shortcuts?